trezor/trezor-firmware #1327: xDai with trezor regression on recent firmwares

yuval-k, 2020-10-27:
summary: I can't use trezor on the xDai side chain with latest firmware; it **does** work with v2.3.0

see description of problem here also: https://www.reddit.com/r/TREZOR/comments/ize369/trezor_xdai/

**Firmware version and revision**
Tested with firmware: v2.3.3, v2.3.4
I can confirm that the problem does not exist in v2.3.0, and transactions on xDai work as expected.

Desktop:
- Trezor Wallet + metamask
- linux
- brave browser

**To Reproduce**
1. Configure xDai rpc in metamask
2. Try to make a transaction in the xDai blockchain (I used honeyswap)

**Expected Result:**
it should work

**Actual result:**
it fails with
"Swap failed: Error: Unsupported chain id"
[image]

# appendix - xdai settings:
Network name: xDai
RPC URL: https://rpc.xdaichain.com/
ChainID: 100
Symbol: xDai
Explorer: https://blockscout.com/poa/xdai/

matejcik, 2020-10-27:
XDAI is not a supported coin per trezor.io/coins, so it's not a bug if it doesn't work

xDAI implementation was piggy-backing on Ethereum support and showing transactions with "ETH" currency symbol, which we consider a potential security risk, and so it is no longer allowed.

closed 2020-10-27 by matejcik

yuval-k, 2020-10-27:
thanks for the quick response @matejcik 
Is there a plan to support xDai in the near term?
I'm asking to know how to proceed - If the answer is no I'll find a different wallet for xDai activities.

matejcik, 2020-10-27:
> showing transactions with "ETH" currency symbol

ah, it turns out I'm wrong here and the transaction was shown with UNKN symbol

matejcik, 2020-10-27:
> Is there a plan to support xDai in the near term?

we will accept a contribution to add xDai support, see https://github.com/trezor/trezor-firmware/pull/1326 for an example of what is necessary