trezor/trezor-firmware #1334: Firmware update cancel  T vs. One inconcistencies

mroz22, 2020-10-30:
How to reproduce:

1. Init firmware update
1. Cancel on device
1. Observe inconcistencies between models

Model T	

```
{
  "type": "Failure",
  "message": {
    "code": "Failure_ActionCancelled",
    "message": "Firmware install cancelled"
  }
}
```

Model One
```
{
  "type": "Failure",
  "message": {
    "code": "Failure_FirmwareError",
    "message": null
  }
}
```

These are objects parsed by trezor-link. Inconcistencies:
- Same action on device, two different codes: `Failure_ActionCancelled`, `Failure_FirmwareError`
- Model One does not send message

tsusanka, 2020-11-06:
This goes from the bootloader right? :-/

mroz22, 2020-11-08:
> This goes from the bootloader right? :-/

Yes. We can live with it, but it is suboptimal (see the linked trezor-connect PR)

prusnak, 2020-11-14:
PR in https://github.com/trezor/trezor-firmware/pull/1347

closed 2020-11-18 by prusnak