trezor/trezor-firmware #1335: Looser checks for replay-protected Ethereum networks

matejcik, 2020-10-30:
With the introduction of strict keychain namespacing, Trezor T stopped working with several not officially supported Ethereum-based networks: see #1326, #1332, #1331 

We believe it is safe to explicitly re-enable support for such networks in the future. There is strong replay protection in the form of EIP-155 chain ID, we use currency symbol `UNKN`, and we can display a warning about "Unknown chain ID" or "Unknown network", whichever seems more user-friendly.

The only open issue is what SLIP44 identifier to use for such networks. We do not want to allow just any, and we can't use chain id in place of SLIP44 because there are mismatches in practice.

One possibility is always using `60`, which is Ethereum id, and the networks often use it in practice. (fwiw, #1184 allows this explicitly, because some networks want cross-signing and token compatibility)
Another is always showing path warning regardless of provided path.

My1, 2020-11-02:
> Another is always showing path warning regardless of provided path.

this may get chaotic fast when people have strict path checks on **which is the default**, like is it supposed to block there and encourage users to get prompts or disable path checks altogether lowering the security?

or should it create a prompt instead thereby creating surprising behavior, which doesnt help users which are confused about enough things already.

in fact a path warning regardless of the setting would be confusing as they would think they are on the "right" path.

https://en.wikipedia.org/wiki/Principle_of_least_astonishment

it should probably rather just have its own warning.

closed 2020-11-20 by tsusanka

bosomt, 2020-12-29:
@matejcik is there anything that can/need to be tested by QA dept. ?

matejcik, 2021-01-04:
@bosomt it would be nice to have this tested, presumably through MEW, but we'd need to find an unsupported Ethereum-like network. And we also added support for all the major ones, so :woman_shrugging: 

i don't even have a good idea where to go looking.

matejcik, 2021-01-04:
Maybe a good idea is to test EWC with derivation path `44'/60'/0'` and `44'/246'/0'`, both should work without warnings.
This is already integrated in MyCrypto, but I'm not sure how to go about changing the derivation path, as I never used the application

bosomt, 2021-01-04:
@matejcik 

I was able to derive 44'/246'/0' addresses without any problem/warning via mycrypto.
BUT it worked with 2.3.4 public version too 

**2.3.5**
[image]

**2.3.4 - both paths**

[image]

[image]

mycrypto beta

[image]

matejcik, 2021-01-04:
> I was able to derive 44'/246'/0' addresses without any problem/warning via mycrypto.
> BUT it worked with 2.3.4 public version too

you would need to send a transaction, the `EthereumGetAddress` call does not differentiate between networks

also this should work for both `44'/246'/0'` and `44'/60'/0'`

commit abc45f98 (2020-11-12): feat(core): re-allow unknown Ethereum networks, using Ethereum or Testnet paths (fixes #1335)
commit d566f657 (2020-11-12): feat(core): re-allow unknown Ethereum networks, using Ethereum or Testnet paths (fixes #1335)
commit 71cdbbfc (2020-11-13): feat(core): re-allow unknown Ethereum networks, using Ethereum or Testnet paths (fixes #1335)
commit cb402e93 (2020-11-20): feat(core): re-allow unknown Ethereum networks, using Ethereum or Testnet paths (fixes #1335)