trezor/trezor-firmware #1336: Forbidden Key Path Error

terryalex64, 2020-10-30:
I tried recovering ETH that I accidentally sent to my Trezor ETC address. I tried using Mycrypto to send to my ETH address and got a Forbidden Key error and when I tried MEW I got the error below. The error happens when hitting send. I do not get any confirmation on the Trezor device. My Trezor Model T has the latest firmware and I had the same problem with the previous firmware.  
{
  "exception": {
    "values": [
      {
        "type": "Error",
        "value": "Forbidden key path",
        "stacktrace": {
          "frames": [
            {
              "colno": 62067,
              "filename": "https://www.myetherwallet.com/js/vendors.89b04c53.js?__WB_REVISION__=933aa79074293eba9beb",
              "function": "s",
              "in_app": true,
              "lineno": 750
            },
            {
              "colno": 61856,
              "filename": "https://www.myetherwallet.com/js/vendors.89b04c53.js?__WB_REVISION__=933aa79074293eba9beb",
              "function": "n",
              "in_app": true,
              "lineno": 750
            },
            {
              "colno": 13801,
              "filename": "https://www.myetherwallet.com/js/vendors.89b04c53.js?__WB_REVISION__=933aa79074293eba9beb",
              "function": "Generator.next",
              "in_app": true,
              "lineno": 807
            },
            {
              "colno": 14700,
              "filename": "https://www.myetherwallet.com/js/vendors.89b04c53.js?__WB_REVISION__=933aa79074293eba9beb",
              "function": "Generator._invoke",
              "in_app": true,
              "lineno": 807
            },
            {
              "colno": 13374,
              "filename": "https://www.myetherwallet.com/js/vendors.89b04c53.js?__WB_REVISION__=933aa79074293eba9beb",
              "function": "l",
              "in_app": true,
              "lineno": 807
            },
            {
              "colno": 1516210,
              "filename": "https://www.myetherwallet.com/js/app.0612e09f.js?__WB_REVISION__=0f5c4ebf58f6fbdbbadf",
              "function": "?",
              "in_app": true,
              "lineno": 1
            }
          ]
        },
        "mechanism": {
          "handled": true,
          "type": "generic"
        }
      }
    ]
  },
  "level": "error",
  "event_id": "18b7fd77a2684643ad1715bb8d5e4f6a",
  "platform": "javascript",
  "sdk": {
    "name": "sentry.javascript.browser",
    "packages": [
      {
        "name": "npm:@sentry/browser",
        "version": "5.21.1"
      }
    ],
    "version": "5.21.1",
    "integrations": [
      "InboundFilters",
      "FunctionToString",
      "TryCatch",
      "Breadcrumbs",
      "GlobalHandlers",
      "LinkedErrors",
      "UserAgent",
      "Vue"
    ]
  },
  "timestamp": 1598303411.8321521,
  "environment": "web",
  "release": "5.7.6",
  "request": {
    "url": "https://www.myetherwallet.com/interface/send-transaction",
    "headers": {
      "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.125 Safari/537.36"
    }
  },
  "tags": {
    "network": "ETH",
    "service": "etherscan.io",
    "walletType": "trezor"
  }
}

My1, 2020-11-01:
this is intentional to prevent malicious applications from making it seem like you would be spending one coin, but actually spend another, as some coins are using exactly the same tech like for example btc and ltc where one couldnt distinguish a btc from an ltc transaction just by themselves, so the derivation paths were locked down to prevent this from happening.
you can soften the path checks with
`trezorctl set safety-checks prompt`
and later lock them back with
`trezorctl set safety-checks strict`

closed 2020-11-02 by matejcik

terryalex64, 2020-11-02:
How do I use this command?  trezorctl set safety-checks prompt
Where do I input it?  Thank you

My1, 2020-11-02:
Considering you asked this I'll just assume you use windows.
https://wiki.trezor.io/Installing_trezorctl_on_Windows

bartlanz, 2021-05-08:
@my1 Dude you saved my transaction with that one. I sent my ETH to my ETC address on my Trezor. I couldnt get it to go. THANK YOU SOOOO MUCH! If I could find a way to contact you directly I would. But thank you thank you thank you.

My1, 2021-05-08:
@bartlanz there's a website in my profile with a contact form, that contacts me directly if you need that.

kelvinwop, 2022-04-15:
Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\trezorctl.exe\__main__.py", line 4, in <module>
  File "C:\Python310\lib\site-packages\trezorlib\cli\trezorctl.py", line 191, in <module>
    @cli.resultcallback()
AttributeError: 'TrezorctlGroup' object has no attribute 'resultcallback'. Did you mean: 'result_callback'?

matejcik, 2022-04-16:
@kelvinwop in case you're not a spambot, here's the issue you are encountering: #2199