trezor/trezor-firmware #1338: Add BSC and BEP20 TWT

prusnak, 2020-11-04:
More info:
* https://github.com/ethereum-lists/tokens/pull/435
* https://github.com/ethereum-lists/chains/blob/master/_data/chains/56.json

matejcik, 2020-11-05:
should we use the chains repo to autogenerate our Ethereum networks?

prusnak, 2020-11-05:
> should we use the chains repo to autogenerate our Ethereum networks?

I think that's a good idea

matejcik, 2020-11-06:
what is the SLIP44 ID for binance?
are they using the same 714 for this and their native chain? i don't think that's a particularly good idea

matejcik, 2020-11-06:
they also seem to use the same BNB symbol. not that great either, seeing as there's currently no way for the user to distinguish between the network they're transacting on

matejcik, 2020-11-12:
BNB smartchain is added in #1345 

we are waiting for https://github.com/ethereum-lists/tokens/pull/435, this is waiting for Trustwallet's modifications

Hannsek, 2023-11-20:
We already have the data from coingecko, so the token is now supported.

closed 2023-11-20 by Hannsek