trezor/trezor-firmware #1343: Explicitly confirm turning off replacement possibility during a replacement

tsusanka, 2020-11-10:
We currently do not confirm the transaction "replacement possibility" (also called RBF sometimes). Neither during the initial sign nor during a replacement process. We have discussed this in Product and we would like to keep the initial signing as is (without any confirmation) not to confuse users, but we would like to have an explicit dialog when the user tries to "finalize" the transaction (removing the replacement possibility).

The particular wording is to be defined by @matejzak when he completes the design in Suite.

andrewkozlik, 2020-11-18:
@matejzak or @tsusanka please clarify some details. I can see four scenarios:
1. If the original transaction **allows RBF** and the replacement transaction **allows RBF**, then keep dialogs as they are currently.
2. If the original transaction **allows RBF** and the replacement transaction **is final**, then show the new finalization dialog.
3. What if the original transaction **is final** and the replacement transaction **allows RBF**? This might seem like nonsense, but the original might have never been sent to the network or it might have been kicked out of the mempool for example due to timeout or due to mempool size limit. So I think technically it is legit. The options are to:
    a) Show no additional dialog.
    b) Show an "unfinalization" dialog.
    c) Fail signing.
4. Similarly, what if the original transaction **is final** and the replacement transaction **is final**? Again might seem like nonsense, but technically it is legit for the same reasons as above.
    a) Show no additional dialog.
    b) Fail signing.

tsusanka, 2020-11-19:
1. Yes.
2. Yes.
3. Failing signing. You can still sign a new transaction no need for the replacement workflow.
4. Failing signing.

We can discuss the details on standup.

andrewkozlik, 2020-11-19:
Based on what we discussed at the standup, we could have the following screens in Case 2 (finalization):

[image] [image]

and if there is no fee change, then the following (ignore that the fee amounts are different):

[image] [image]

In Case 1 (no finalization) we would keep the current screens:

[image] [image]

In Cases 3 and 4 I am inclining towards showing allowing them and showing no additional dialog at least in the PayJoin scenarios, which would keep looking like this if the user's fee contribution is increasing:

[image] [image]

or simply like this if the *user's* fee contribution is decreasing:

[image]

andrewkozlik, 2020-11-19:
I just realized another reason why allowing Cases 3 and 4 makes sense. Even if an original transaction appears finalized, it may have inherited RBF opt-in if one of its ancestors opted-in to RBF and remains unconfirmed, which we can't detect on Trezor.

closed 2020-11-20 by tsusanka