trezor/trezor-firmware #1357: Trezor v2 bootloader: Add decred vendor pubkeys.

JoeGruffins, 2020-11-20:
**Is your feature request related to a problem? Please describe.**
Decred would like to supply it's users with custom firmware, but this creates a security problem if we must use the "unsafe" developer header because anyone can slip in malicious firmware after installing our firmware without a seed wipe.

**Describe the solution you'd like**
Add decred supplied public keys to the bootloader that we can sign with our private keys.

**Describe alternatives you've considered**
Add changes to master. This takes exactly one eternity and we will all be dead before it takes place. (Joking, please don't take offense)

**Additional context**
none

prusnak, 2020-11-20:
That's not how it works. You are supposed to create a vendor header and send it to us to have it signed. Then you attach the signed vendor header to your firmware.

closed 2020-11-20 by prusnak

JoeGruffins, 2020-11-20:
That would be a great solution! Where can I send the vendor header?

prusnak, 2020-11-20:
Send me an email to stick@satoshilabs.com and we'll continue the discussion there