trezor/trezor-firmware #1366: FIDO2 PAM does not work

NullSense, 2020-11-23:
**Describe the bug**
```
$ sudo pamu2fcfg
error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT
```

**Firmware version and revision**
Firmware: 2.3.4
Bootloader: 2.0.3

**Expected behavior**
Appropriate pamu2fcfg output to be used for PAM auth.

prusnak, 2020-11-23:
What is the version of pamu2fcfg? (`pamu2fcfg --version`)

tsusanka, 2020-12-07:
Ping @NullSense

prusnak, 2020-12-07:
@NullSense please also provide libfido2 version

NullSense, 2020-12-07:
1.1.0 and 1.5.0-3 respectively. (Arch linux)

NullSense, 2020-12-07:
@tsusanka @prusnak

prusnak, 2020-12-08:
I confirm that I can reproduce the issue with the following setup:
* pamu2fcfg 1.1.0
* libfido2 1.5.0
* trezor-firmware 2.3.4

prusnak, 2020-12-08:
The same setup works with Trezor One which does not have FIDO2 support, so only U2F is used.

Contents of the packets can be inspected via `pamu2fcfg -d`.

My wild guess is this might be connected to self-attestation, which might not be correctly handled by pamu2fcfg.

andrewkozlik, 2020-12-09:
@prusnak is absolutely correct. `pamu2fcfg` does not support self attestation. So the options are:

1. Patch `pamu2fcfg` to support self-attestation, unless it's intentional that it doesn't support it.
    -  The place to apply the patch is https://github.com/Yubico/pam-u2f/blob/5b504783c9af972c790bdcb506867bad7df5e333/pamu2fcfg/pamu2fcfg.c#L288-L292, with something like this:
 ```
    if (fido_cred_x5c_ptr(cred) == NULL) {
        r = fido_cred_verify_self(cred);
    } else {
        r = fido_cred_verify(cred);
    }
```
2. Use basic attestation in Trezor for RP IDs beginning with "pam://".
    - We would need to add wildcard or regexp support to the RP ID matching in Trezor, which might come in handy in the future anyway.
    - It's not a full solution, because the user can specify any RP ID using the `-o` option: `pamu2fcfg -o RP_ID`. So if someone specifies an RP ID not beginning with "pam://", then it won't work.
3. Switch to using basic attestation by default in Trezor.
    - Seems OK, I think we use self-attestation by default just because it's more "honest".

andrewkozlik, 2020-12-09:
https://github.com/Yubico/pam-u2f/commit/03ebe095995b9e250e6c523b066b04eaf78907cd

andrewkozlik, 2020-12-09:
The [patch](https://github.com/Yubico/pam-u2f/pull/168) was just merged into `pam-u2f`, so this issue should be fixed in the next release of `pamu2fcfg`.

closed 2020-12-09 by andrewkozlik