trezor/trezor-firmware #1368: Legacy account send transaction fail - Required field amount was not received

wendys-cats, 2020-11-24:
**Describe the bug**
This bug affects only Trezor Wallet Legacy transaction (send from legacy)  when using TT 2.3.5. Wallet does not display the same behavior with T1 1.9.3 (public) nor TT 2.3.4 (public), nor does the same bug occur in Suite. 

Occurs for following coins: BTC, LTC.

When sending transaction from a Legacy account, an error occurs with following message: "Error details: Failed to decode message: Required field 'amount' was not received"

**Firmware version and revision**

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Wallet 98bf3eb12c1d85029310448e24ba4d88b312cb7c
 - OS: NixOS 21.03
 - Browser Chromium 86
 - Version TT 2.3.5 c28a577 regular

**To Reproduce**
Steps to reproduce the behavior:
1. Go to BTC/LTC Legacy Accounts
2. Fill Send form
3. Click on Send
4. See error

**Expected behavior**
Transaction will be sent.

**Screenshots**
[image]

[image]

**BTC**
[image]

**LTC**
[image]

andrewkozlik, 2020-11-24:
This sounds like https://github.com/trezor/trezor-firmware/issues/1109 and should be fixed by https://github.com/satoshilabs/mytrezor/pull/949.

andrewkozlik, 2020-11-24:
So I checked and Trezor Wallet 98bf3eb12c1d85029310448e24ba4d88b312cb7c is the current `develop` and `stable/2020-11-24`, but it does not include my commits from https://github.com/satoshilabs/mytrezor/pull/949. So this behavior is to be expected. The firmware should work correctly with https://mytrezor.corp.sldev.cz/input-amounts-always/development/#/ as confirmed by QA in #1109. @tsusanka @trezor/qa I think there is no point in testing Trezor Wallet until https://github.com/satoshilabs/mytrezor/pull/949 is merged.

tsusanka, 2020-11-24:
Yes, this is expected. Apologies, I forgot to warn you about this. I will instruct @mroz22 to merge satoshilabs/mytrezor#949 tomorrow.

> I think there is no point in testing Trezor Wallet until satoshilabs/mytrezor#949 is merged.

I disagree, there is a lot of other things to test. We will test BTC signing again after it is merged though.

closed 2020-11-24 by tsusanka

tsusanka, 2020-11-26:
@wendys-cats just to make sure could you please retest at https://mytrezor.corp.sldev.cz/develop/development/ ?

wendys-cats, 2020-11-26:
**QA OK**

**OS:** NixOS 21.03
**Browser** Chromium 86
**Version** TT 2.3.5 c28a577 regular
**Wallet** [develop](https://mytrezor.corp.sldev.cz/develop/development/) branch

**Legacy to legacy** [transaction](https://btc2.trezor.io/tx/c062d6f05bf59906ec22dd31b61f1aaa02db416504accd915d31caea71b49c1f)
**Legacy to segwit** [transaction](https://btc2.trezor.io/tx/c0332deff729f66c4526efbda4979ca7cc0476358f91c2fd32fb036f5354121d) 
**Legacy to bech** [transaction](https://btc2.trezor.io/tx/e9e927d36197c31d9842d360d2c6427ccd0eac72744de17f77a3f2205da1992a)

tsusanka, 2020-11-26:
Thanks. Could you please do one more with 2.3.4 on the very same URL? Just to make sure we did not break anything.

wendys-cats, 2020-12-01:
**QA OK for 2.3.4**

**Wallet** [develop](https://mytrezor.corp.sldev.cz/develop/development/) branch 
**OS:** NixOS 21.03
**Browser** Chromium 86
**Version** TT 2.3.4 public regular

**Legacy to legacy** [transaction](https://btc2.trezor.io/tx/e30bd0a97767daad55e1eed8aef661a09dae072d899138872c0e47d6f540ae57)
**Legacy to segwit** [transaction](https://btc2.trezor.io/tx/1107cec02027e92c028f6a7951d5ed5bc606cff2640b2a03766136088b2c40cd)
**Legacy to bech** [transaction](https://btc2.trezor.io/tx/9e1d79564a23a6d94706edf0d308a2b8b00e660d31304b40da8a7cf8d524c688)