trezor/trezor-firmware #1371: TestMultisig::test_15_of_15 fails on device due to fragmented heap

mmilata, 2020-11-26:
**Describe the bug**
While there is multiple places where I've seen the device returnin an error caused by fragmented memory, I chose to write down what I found out about this one because it is easy to trigger reliably.

* On normal firmware in debug mode it suffices to run the test right after boot for it to fail.
* On btconly firmware in debug mode you need to run couple of other tests before this one for the bug to manifest:

```
28209056 trezor.wire ERROR exception:
Traceback (most recent call last):
  File "trezor/wire/__init__.py", line 387, in handle_session
  File "trezor/loop.py", line 526, in __iter__
  File "trezor/loop.py", line 191, in _step
  File "apps/bitcoin/keychain.py", line 184, in wrapper
  File "apps/bitcoin/sign_tx/__init__.py", line 70, in sign_tx
  File "apps/bitcoin/sign_tx/bitcoin.py", line 58, in signer
  File "apps/bitcoin/sign_tx/bitcoin.py", line 194, in step4_serialize_inputs
  File "apps/bitcoin/sign_tx/bitcoin.py", line 576, in sign_nonsegwit_input
  File "apps/bitcoin/sign_tx/bitcoin.py", line 728, in input_derive_script
  File "apps/bitcoin/scripts.py", line 68, in input_derive_script
  File "apps/bitcoin/scripts.py", line 406, in input_script_multisig
  File "apps/common/writers.py", line 12, in empty_bytearray
MemoryError: memory allocation failed, allocating 1602 bytes
```

Allocations for the single test: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/875883108/artifacts/core/prof/memperf-html/index.html

**Firmware version and revision**
2020-11-26 master / 01d105804821c881b24a2d69b58e6094eb1f5f2c

mmilata, 2021-02-17:
Also happens with `test_msg_signtx_segwit.py`, see #1479. This wasn't the case before f1382bf.

mmilata, 2021-02-18:
Also `test_op_return.py`. See `$TESTS_SKIP` in [`ci/test-hw.yml`](https://github.com/trezor/trezor-firmware/blob/master/ci/test-hw.yml) for more complete list.

mmilata, 2021-06-21:
Update: there was a massive improvement in heap fragmentation thanks to #1541 and #1565, however `test_15_of_15` remains one test that we still skip.

andrewkozlik, 2021-07-22:
#1581 is now merged into master. Based on https://github.com/trezor/trezor-firmware/pull/1581#pullrequestreview-712152179 it should be OK to enable `test_15_of_15` and close this issue.

closed 2021-10-07 by mmilata