trezor/trezor-firmware #1373: Add support for Wrapped ILCoin ERC-20 token

PlusBitPos, 2020-11-27:
I checked **Developers guide: Contributing**; under the title "**Adding a new coin**", it forwards me to
https://github.com/ethereum-lists/tokens (Ethereum tokens should be added to the ethereum-lists/tokens repository).

I made a PR there https://github.com/ethereum-lists/tokens/pull/441, but it seems this repo belongs to different account not trezor. 

I think Trezor beta wallet already supports WILC because it's an ERC20 token. **My aim is to add the WILC to the list of your supported coins (https://trezor.io/coins/) and to the Token list in your beta wallet.** 

If there is something extra that you will need in order to add our token, please let me know. Maybe just the logo in order to differentiate the token with more than just the name. I provided all related data below. 

**symbol**: WILC,
**name**: Wrapped ILCOIN,
**type**: ERC20,
**address**: 0xc98a910ede52e7d5308525845f19e17470dbccf7,
**decimals**: 8,
**logo**: https://raw.githubusercontent.com/PlusBitPos/PlusBit-graphic-kit-for-media-and-graphic-identity/master/icons/logo-512.png,
**etherscan**: https://etherscan.io/token/0xc98a910ede52e7d5308525845f19e17470dbccf7

prusnak, 2020-11-27:
Once the PR in the tokens repo is accepted the token will eventually make it into our firmware (in the next firmware update).

closed 2020-11-27 by prusnak

PlusBitPos, 2020-11-30:
Thank you very much Dear @prusnak 
@ligi  merged PR; https://github.com/ethereum-lists/tokens/pull/441
for now, i only need to wait next firmware update without doing anything to add the WILC token to the list of your supported coins (https://trezor.io/coins/) and to the Token list in your beta wallet?
Kind regards

PlusBitPos, 2021-02-14:
Hello @prusnak, sorry for disturb,
First of all thank you for new update, 
However, it seems this version does not include new erc20 token from ethereum-list/tokens repo. Although ligi merged the PR in the tokens repo, There might have been some issue with WILC? It will included in next update or from now trezor only allow for manual addition by contract address?
Looking forward your help, 
Best regards

prusnak, 2021-02-14:
We pull the new definitions and release a new firmware in March

PlusBitPos, 2021-03-10:
@prusnak hi, sorry for disturb,
I am currently running version 21.3.1 released today. Also checked the beta wallet (https://beta-wallet.trezor.io/next/#/). However WILC token is not there. It seems you missed it for this update too. 
[image]

on the other hand, there is an issue with trezor suite, the "add token" button stays passive while adding the token manually.
[image]

prusnak, 2021-03-10:
1. There was no firmware update this month. You need to wait for the next month's firmware release.
2. Don't use beta wallet.
3. Please report the Suite issue to the suite repo: https://github.com/trezor/trezor-suite