trezor/trezor-firmware #1375: Revise reward transactions UI

andrewkozlik, 2020-11-30:
1. In Komodo reward transactions we show the fee amount as negative in `core` and as 0 in `legacy`. Apparently neither of these is correct. I am not sufficiently familiar with Komodo, but I think that to show the correct fee we would need to know the reward amount. The transaction in `test_msg_signtx_komodo.py-test_one_one_rewards_claim` has the reward as a separate output. Is that always the case? If so, is there a way to identify the reward output? Perhaps the total amount of rewards could be sent as a separate piece of information in `sign_tx` and we would show the user an additional dialog saying "Claim reward of ... KMD?" after confirming the last output, and then include the reward amount in the fee computation. If we cannot show the correct fee, then Trezor should display "including fee: unknown" or something similar.

2. I think the same problem applies to the total amount shown in the final confirmation screen. The amount being spent in a transaction with rewards is larger than the amount shown, because the amount shown does not include the rewards. Furthermore, if the outputs went to change-addresses, then the total amount would come out negative. On T1 we would show `(uint64_t)total_in - change_out` as the total amount, which wraps to billions of KMD.

[image] [image]

andrewkozlik, 2020-11-30:
Note that this might also have implications for showing the fees in replacement transactions. In T1 I decided to reject replacement transactions which involve negative fees in the original or replacement to avoid dealing with the UI implications. Maybe we should do the same in trezor-core.

matejcik, 2020-11-30:
If the Komodo network allows such transactions, we have no basis to determine what the transaction data really mean. In particular, if the fee comes out positive, but rewards are being spent, we won't even know this is happening.

As this is a feature of a coin not supported in Suite, I don't think we need to spend effort on this. (even if we wanted to, this would most likely need cooperation from the host software, which is outside our control).

I think we should fix the T1 issue with negative numbers, and after that, the TT info is good enough.

FWIW there is a vaguely similar issue with OMNI transactions, where the displayed info seems also technically correct but missing so much context as to be basically useless. That is something that will need some work, but only if/when we add Tether to Suite.

> Maybe we should do the same in trezor-core.

agreed

andrewkozlik, 2020-12-01:
To summarize there are four problems:

1. T1, T2: The total amount we show is not really "total", because it doesn't include the reward which is being spent too.
**Resolution:** Ignore for reasons @matejcik gave.

2. T1, T2: The fee we show is not really the fee. What we show as fee is `fee - reward`.
**Resolution:** Ignore for reasons @matejcik gave.

3. T1: Total amount breaks for negative numbers.
**Resolution:**
a) Show negative value like we do on T2.
b) Show 0 when total is negative (like we do with the fee on T1).

4. T1 vs T2: Negative fee shows up differently (see screenshots).
**Resolution:**
a) Show negative value on T1, like we do on T2.
b) Show 0 on T2, like we do on T1.
b) Keep on showing it differently on the two models.

Anybody else care to weigh in, @tsusanka, @prusnak, @matejzak ?

tsusanka, 2020-12-07:
As discussed on standup: 3.A 4.A.

andrewkozlik, 2020-12-09:
I added the fix to the T1 replacement transactions PR https://github.com/trezor/trezor-firmware/pull/1380 in https://github.com/trezor/trezor-firmware/pull/1380/commits/fabd4ebdb9f3f4811965c0f117f776783b10b61d. This is the result:
[image]

tsusanka, 2021-10-05:
Hasn't #1380 closed this one?

closed 2021-10-05 by andrewkozlik