trezor/trezor-firmware #1385: Documentation outdated

mroz22, 2020-12-14:
There is a link to toif_convert tool on:

https://docs.trezor.io/trezor-firmware/misc/toif.html

[image]

Link takes you to https://docs.trezor.io/trezor-firmware/tools/toif_convert.py which does not exist at the moment. 

Not sure if this is a bug or a feature request :thinking:

prusnak, 2020-12-14:
Fixed in https://github.com/trezor/trezor-firmware/commit/d9a8194e14b31f64fb3a443f4f97acf0b24ef427

(use absolute link instead of relative)

closed 2020-12-14 by prusnak

mroz22, 2020-12-14:
ah it lives here https://github.com/trezor/trezor-firmware/blob/master/core/tools/toif_convert.py

mroz22, 2020-12-14:
thanks!! that was fast