trezor/trezor-firmware #1387: Cardano key derivation deviates on 24-word mnemonic

refi93, 2020-12-15:
**Describe the bug**
Cardano root key derivation for 24-word mnemonics deviates from 12 and 18-word mnemonics respective to other Cardano wallet implementations.

**Firmware version and revision**
Any version since Cardano support has been launched

**Desktop/smartphone setup (please complete the following information):**
n/a

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize trezor with some 24-word mnemonic, e.g. `balance exotic ranch knife glory slow tape favorite yard gym awake ill exist useless parent aim pig stay effort into square gasp credit butter`
2. Go to 'https://adalite.io'
3. Log in with trezor, the first address will be `addr1q8qch787x834kstfkuh6g5q0axyt8jxdwem5cwt8t8tnjhk68e0lartvutmlvf2nra9ehp57w3mtyawca3nfga4pk7nqccykxn`
4. Log in directly with the same mnemonic into adalite and the first address will be `addr1q8m50vdutylp8ju3cwudqy2qj5wh5grh469xgnuvj3vq9s5xttc7l65a369kkq9re6nlxt77kyjs57xy4yw3qlrl3lcqcf6kkq`

**Expected behavior**
For the same 24-word mnemonic the Cardano addresses/keys derived should match, the same as the do for 18-word mnemonics (tried e.g. with `found differ bulb shadow wrist blue bind vessel deposit tip pelican action surprise weapon check fiction muscle this`)

**Additional context**
We investigated the root cause of the bug and apparently, the function that feeds the BIP39 entropy into the function to generate Trezor's Cardano root key takes the entropy from a function that actually returns the entropy + checksum and computes the entropy byte size as the amount of entropy bits divided by 8. This actually works fine for 12 and 18 words as the checksum is 4 and 6 bits respectively, but for 24 it happens to be 8 so one more byte is passed to the function that generates the cardano root key, deviating from the [BIP39 specification](https://github.com/bitcoin/bips/blob/master/bip-0039.mediawiki#generating-the-mnemonic).

Here's Trezor's mnemonic_to_entropy function: https://github.com/trezor/trezor-firmware/blob/d657cdaeef9906ab37d42a72d8e7ed1bd09adc9e/crypto/bip39.c#L97

And here's the offending line of the function that derives the root key: https://github.com/trezor/trezor-firmware/blob/c67ecb984bc3f83eb4f0781eb8d9db3e5c553e91/core/embed/extmod/modtrezorcrypto/modtrezorcrypto-bip32.h#L611

Historically, the bug seems to have been introduced back in the day when we were implementing Cardano into Trezor and we reused the code of the mnemonic_check function() which indeed dealt with the entropy + checksum: https://github.com/trezor/trezor-crypto/commit/e827517591a65b7301ecb629ce5caf816a6f03cc#diff-537ce5c99775eb59008c38bd9c01598d7cff405a4fb861077c33dc7fa906be11R55

Since the reversal of the incorrect behavior would result in an alteration of users' Cardano keys and addresses derived on Trezor if it is initialized with a 24-word mnemonic, we propose at least refactoring the code to obtain the mnemonic entropy to be less misleading (i.e. to actually return the entropy and not entropy+checksum), so future coin implementations don't get "tricked" by this and document/cover with tests the deviation accordingly so it's not accidentally "fixed" in the future. We are already working on such PR.

tsusanka, 2021-01-04:
This is unfortunate but we can't do anything about it since it is already released. #1388 should prevent this from happening again. Closing.

closed 2021-01-04 by tsusanka

Teralodom, 2021-03-13:
@refi93 @tsusanka 
Hello!

Does this mean that my 24 or 18 word seed on Trezor for Cardano can only be used on Trezor?

I saw one post on Reddit of a guy saying that he moved from Ledger to Trezor and he could restore all the coins except Cardano, it yielded a different wallet.

1. So, if I lose my Trezor device, I would need to get another Trezor to be able to retrieve my wallet?
2. From what I understood 12-word mnemonics are safe, right?

Thank you,

refi93, 2021-03-13:
@Teralodom answers to your questions:

1. Unfortunately yes, Trezor's Cardano support is simply different from Ledger, the reason being that Trezor came earlier and it's architecture allowed for implementing Cardano's official scheme of deriving keys, something that later with Ledger turned out to be unfeasible, so that's the main reason of the diffrence. The deviation at 24-word mnemonics is just a contributing factor breaking comaptibility of Trezor even with Cardano software wallets (but inserting your mnemonic into a software wallet is a very bad practice regardless, you should never do that in fact, unless you know what you are doing).

An alternative to getting the physical device could be setting up a Trezor emulator on your PC: https://docs.trezor.io/trezor-firmware/core/emulator/index.html and transferring all the funds (not only Cardano) to another trusted hardware wallet using a different mnemonic ASAP

2. not sure if you mean general safety of 12-word mnemonics - if you do, 12-word mnemonics are generally deemed safe enough as mentioned in Trezor docs as well: https://wiki.trezor.io/Recovery_seed Feel free to clarify your question if you had something else in mind

nargotik, 2021-03-19:
So, this issue is closed this means that cardano on trezor will be in a non standard way !!! How this will be solved, since there shall be many trezor users and putting trezor in a standard way shall put cardano unavailable.

This looks like a scary bug and the way to solve this is to let it be ... 👎

refi93, 2021-03-19:
I agree the bug looks scary and it's unfortunate that it breaks interoperability.

Unfortunately, the implications of reversing that bug would be that users having a 24-word mnemonic on Trezor would not be able to access their Cardano funds stored before the bug has been "fixed". Such change would be justifiable only if there was any actual security risk which to the best of our knowledge there isn't, the rationale being that the entropy, even if longer by 1 byte, is still being hashed at the next step, the hash function being agnostic to the input length and its output doesn't have any structural constraints either except its length.

We documented the deviation and informed IOHK/Trezor about it. So far Trezor's deviation was only acknowledged but not explained in the official Cardano standards, so I added a PR with the explanation: https://github.com/cardano-foundation/CIPs/pull/76/files The Trezor derivation of Cardano keys basically became a standard of its own and it's already officially acknowledged in Cardano's standards as well: https://github.com/cardano-foundation/CIPs/blob/master/CIP-0003/CIP-0003.md ("not recommended" there means just that other wallets should not replicate it if they don't have to, it has no security connotation)