trezor/trezor-firmware #1392: Dealing with mnemonic formats other than BIP39, their ambiguity, and nonstandard derivation paths

andronoob, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
There're competing mnemonic formats other than BIP39. Developers want to "fix (so-called) problems BIP39 by replacing/killing BIP39", which is exactly the ridiculous real example of [xkcd 927](https://xkcd.com/927/).

Even worse, there are "BIP39 killers" who share exactly the same (English) word list with BIP39, like Electrum 2.0 seed and aezeed, which creates ambiguity problem.

BIP44 came too late, later than early BIP39 adopters like BRD, which had been using BIP39 with "nonstandard" BIP32 HD derivation path before BIP44. (if I didn't misunderstand this history)

There are also cases like Bitcoin Core's "simple" HD implementation, which uses hardened derivation (rather than normal derivation) on addresses.

**Describe the solution you'd like**

1. Support importing mnemonic formats other than BIP39. For example, it doesn't seem to be hard to support importing existing Electrum 2.0 mnemonics (rather than both importing existing mnemonics and generating new mnemonics), only hashing and predefined valid version bits (the whole word list is not needed) are required.

2. Properly handle the ambiguity problem - there's a small but still non-negligible probability to generate Electrum2.0/BIP39 ambiguous mnemonics, or something alike. The wallet should be aware of this situation, so that it could prompt the user to choose correct wallet type.

3. Just like [what Electrum had done recently: auto-probing/scanning possible HD derivation paths](https://github.com/spesmilo/electrum/blob/b80978c8db013b71754107bcabceaf9f545975a9/RELEASE-NOTES#L52) - I wish that ambiguous mnemonics can also be covered by this.

4. Educate/warn the user not to create/use wallet/mnemonic in ambiguous ways. Warn users about the danger of exotic ("deep", outside the gap limit) derivation paths.

5. Cooperate with other wallet developers to make the entire ecosystem better.

prusnak, 2020-12-17:
This is out of scope. We suppport BIP39 and SLIP39 mnemonics. There has been no other formats standardized yet and we see no reason to adopt them.

closed 2020-12-17 by prusnak

andronoob, 2020-12-28:
>This is out of scope. We suppport BIP39 and SLIP39 mnemonics.

Maybe. However it's a simple fact that there are popular seed formats which are ambiguous to BIP39. At least there're issues around ambiguity.

>There has been no other formats standardized yet and we see no reason to adopt them at the moment.

Frankly speaking, it's more a social/"political" issue, rather than a pure technical one, so that I wish Trezor could become a good precedence in this ecosystem. (To be honest, I think both seed formats of Electrum 2.0 and aezeed have already set a bad precedence, which encourage developers to design/implement incompatible/fragmentated things, so that users will probably suffer more in the future) Although I think it's not favorable for Electrum to discriminate BIP39 by means like forcing users to enter BIP32 HD derivation path manually (okay, to be fair they only do this in the past, and they still provided a default derivation path), I still appreciate Electrum because they at least provide basic level of support to BIP39, and even better, recently they introduced automatic probing/scanning for unknown derivation paths.