trezor/trezor-firmware #1394: Verify CoinJoin coordinator's URL

tsusanka, 2020-12-18:
In the CoinJoin authorization workflow we display coordinator's URL. As discussed in Product we would like to have this cryptographically signed so Trezor can verify the URL and potentially allow/disallow it. 

[image]

Our general idea is that the coordinator would have some kind of "header" which is signed by SatoshiLabs. What does @andrewkozlik think?

andrewkozlik, 2020-12-21:
Just to explain how this works at them moment: The name of the service provider gets included in the *commitmentData* of the ownership proof(s) generated by Trezor. So the purpose of this screen is to affirm the user's intent to participate in a CoinJoin transaction with the given service provider. This acts as a mitigation against denial-of-service attacks, so that an attacker cannot reuse the proof of ownership to register an input into a different CoinJoin transaction than the one for which the proof was intended.

So currently this confirmation protects the system as a whole against DoS, but doesn't provide any assurance to the user that this is indeed where the CoinJoin will take place. We could add this feature by using the payment request mechanism that I recently proposed. Once we implement that, it seems pretty simple to apply it to CoinJoin. The signed payment request would get sent to Trezor in the final phase of CoinJoin when the transaction is getting signed.

The added value to the user is not exceptionally significant. The current implementation already protects against coin loss. Furthermore Suite will presumably be using a TLS connection with the coordinator, which will give the user assurance about the coordinator's authenticity. So this feature would just give the user a second level of assurance about who the coordinator is. The coordinator's authenticity is certainly something to be concerned about, because the user trusts the coordinator to mix the user's input with the inputs of other honest users. A malicious coordinator could mix the user's input with the coordinator's own inputs only, giving them a false sense of anonymity.

So I agree this is a good idea once we implement signed payment requests.

tsusanka, 2021-02-01:
Will be solved by #1429.

tsusanka, 2021-02-19:
Since we have #1430 for payment requests I do not think we need this one.