trezor/trezor-firmware #1398: TrezorCtl SD card FAT FS Read / Write / List implementation request.

brianddk, 2020-12-23:
**Is your feature request related to a problem? Please describe.**
This is a request for a new feature and and only provides convince and ease of use.

The goal is to provide a very then file read/write capability to the user through Trezor FW to the SD card while it is connected.  A thin implementation like `scp` would likely be much narrower in scope than a generic `usbstor` module / driver.

**Describe the solution you'd like**
A new FW messages to write a blob to the SD card and read a blob from the SD card and list/read a directory table.  One of the parameters could allow the read / write operation to happen either with or without EncFS support.  If an EncFS option is implemented then there is a large amount of work to implement `encfs` in firmware.  The master key for an `encfs` IO would of course be derived from the Trezor keystore.

The best place to expose these operations would be through `trezorctl`.  Something like:

### Read / Write / List without encryption
```
trezorctl sd list usb:/path
trezorctl sd copy c:\path\file.ext usb:/path/file.ext
trezorctl sd copy usb:/path/file.ext c:\path\file.ext 
```

### Read / Write / List with EncFS encryption
```
trezorctl sd list usb:/path
trezorctl sd copy --encfs c:\path\file.ext usb:/path/file.ext
trezorctl sd copy --encfs usb:/path/file.ext c:\path\file.ext 
```

Obviously the encryption would only be for reads / writes to the `usb:\` volume.

**Describe alternatives you've considered**
Since the SD card is easily removable this is likely just window dressing, but may still be a nice convince.  It may provide a later functionality by chaining with other features that I haven't outlined here.

One such extension may be an in-app file browser using Trezor Suite.   Users could launch a sandboxed file browser that would allow drag and drop capabilities that would all be powered by the lower level list/read/write messages.

**Additional context**
Reddit thread with @matejcik 

* https://www.reddit.com/r/TREZOR/comments/kicqxm/-/ggq6545/

brianddk, 2020-12-23:
As an after thought.  This is definitely something that would need to be opt-in with an on-device prompt.

```
# trezorctl sd --enable-io
Please confirm this action on-device
```

OiX0ufie, 2020-12-23:
I really like this idea. Maybe also add a "remove" interface - together with "list" and "copy" the basic command set should be complete.

stepmes, 2023-02-13:
Hi @brianddk, I'm closing this issue, but we'll still keep it internally together with all other user feedback from various sources. At the user research team, we'll make sure we've it on hand whenever topics around the SD card will be discussed in the future. Thanks for sharing your idea, and all the best!