trezor/trezor-firmware #1399: No device label character limit in trezorctl

bosomt, 2020-12-27:
**Describe the bug**
There is no character limit in trezorctl  and device recovery  in Electrum and you can set or try to set device label as long as you want.
omdel One end with `Error: DataError: string overflow` and model T long label is applied.
Some apps like Suite are not expecting it and it's breaking it's grafics

**Firmware version and revision**
Model T 2.3.5 commit 889a369
Model One 1.9.4 commit 889a369

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezorctl 0.12.2
 - OS: Nixos
 
**To Reproduce**
Steps to reproduce the behavior:
1. Set model T device label via trezorctl
2. Check device screen or go to https://suite.trezor.io/web/ and observe

**Expected behavior**
There is same character limit as in Suite and wallet.

**Screenshots**
[image]
[image]

**Note(s):**

**Model One**
trezorctl set label 12345678912345678912345678933333333333333333333333333333355555555555555555555555555555555555555555DDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDD
WebUSB transport is disabled: libusb-1.0.so: cannot open shared object file: No such file or directory
Error: DataError: string overflow
(trezor-firmware)

**Model T** 
[tbosansk@tbosansk:/home/tbosansk/Downloads/trezor-firmware]$ trezorctl set label 12345678912345678912345678933333333333333333333333333333355555555555555555555555555555555555555555DDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDD
WebUSB transport is disabled: libusb-1.0.so: cannot open shared object file: No such file or directory
Please confirm action on your Trezor device.
Settings applied
(trezor-firmware)

prusnak, 2020-12-27:
For reference: label in the legacy firmware is limited to 32 characters (33 bytes of zero-terminated string)

matejcik, 2021-01-12:
@bosomt please re-test on master?

closed 2021-01-12 by matejcik

matejcik, 2021-01-12:
reopening for `Error: DataError: string overflow` on T1
not sure if that is worth fixing, will discuss

bosomt, 2021-01-29:
**QA OK**

**info**
Nixos
FW 2.3.5   model T  revision: 9 bytes b'ffa96205f',
trezorctl version 0.13.0

commit 38ee4371 (2021-01-11): feat(core): introduce and enforce limit on label length (fixes #1399)
commit e3dcf402 (2021-01-11): feat(core): introduce and enforce limit on label length (fixes #1399)
commit fc4e15fe (2021-01-12): feat(core): introduce and enforce limit on label length (fixes #1399)