trezor/trezor-firmware #1404: Add option to lock the device from the screen on TT

tsusanka, 2021-01-04:
Before https://github.com/trezor/trezor-firmware/issues/61 is tackled we would like to add at least the option to lock the device. The idea is that if you hold your finger on the homepage screen you will be prompted whether you indeed want to lock the device. Eventually in #61 the hold will be replaced with some menu.

It is the thing same as T1 has when you hold left button for 5s. Let's use the same delay.

STew790, 2021-06-02:
QA OK

Pressing the screen for 2.5s locks the device.

**Info:**
 - Suite version: web 21.6.1 (7d50da6b9fe39ddd934fdf6153b4c4afb2d7fd88)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.93 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: model T 2.4.0 071c6c0

STew790, 2021-06-02:
QA OK

Holding a finger on screen locks the device.

**Info:**
 - Suite version: web 21.6.1 (7d50da6b9fe39ddd934fdf6153b4c4afb2d7fd88)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.93 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: model T 2.4.0 071c6c0

closed 2021-06-02 by STew790

wendys-cats, 2021-06-02:
**QA OK**

PIN needs to be enabled.
Pressing a finger on screen tiggers locking for TT.
Holding the left button triggers locking for T1. 

**Info:** 
- Suite version: web 21.6.1 (7d50da6b9fe39ddd934fdf6153b4c4afb2d7fd88)
- Browser: Chromium Version 88.0.4324.182 (Official Build) (64-bit)
- OS: NixOS 21.03
- Device: TT 2.4.0 071c6c0 and

PR #1422 (2021-02-23): Lock TT by holding finger on home screen
Fixes #1404. The usual hold-to-confirm animation is shown after first 500ms and finishes in 5s in total. The PR also extends debuglink to support touch hold events.

Please test on actual device as the overall impression can be very different from emulator:)
([regular debug build](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/972529766/artifacts/file/trezor-fw-regular-debug-2.3.6-6ce0745b.bin))

[image]