trezor/trezor-firmware #1407: Marcineczek

MGmarcineczek, 2021-01-06:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware version and revision**
Describe firmware version and revision you are using when encountering the bug.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Wallet, Trezor Beta Wallet, Trezor Suite, Electrum]
 - OS: [e.g. Windows, macOS, Linux, Android]
 - Browser [e.g. chrome, firefox, safari]
 - Version [e.g. 42]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

MGmarcineczek, 2021-01-06:
Where are can interesting use my ekont

closed 2021-01-06 by prusnak

MGmarcineczek, 2021-01-07:
J hawe problem for all my works j don can send my money for wallet and can
not use my money.
Give me answer why are can not use.

prusnak, 2021-01-07:
Github is not a support channel.

Please contact our support via https://trezor.io/support/technical/issue/

MGmarcineczek, 2021-01-15:
Nice weekend we can lettel rest and give Ace rest too mirrow. As job is not easy but said when do t be hawe help from frends c a n not be live be that hight be. All still going will be  next pokolenie mbeda   już all atomatuc and