trezor/trezor-firmware #1408: Shamir backup for passphrase

petrhejna, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
Forgetting or loosing the passphrase results in complete loss if funds. It seems that it may be good idea to backup it. Because it is key part of information for generating a private key I would like to threat it the same way as seed is.

**Describe the solution you'd like**
- I like to be able to do Shamir Backup for a passphrase as a standard TREZOR feature.
- Backup should be somehow padded (or any other solution) to allow multiple passphrases (for example 5 of themem) preserving the plausible deniability. (Someone holding one share should not be able to determine how many passphrases I have).

**Describe alternatives you've considered**
- Shamir backup can be done "manually" (for example using http://point-at-infinity.org/ssss/) but it suffers from lack of standardization.

**Additional context**
- Writing passphrase on piece of paper / into steep, duplicating it with every shard of a seed shamir backup. This bad, because anyone holding a one shard can now steel my funds by just physically stealing my TREZOR and forcing me to reveal PIN (without possible plausible deniability of having multiple passphrases).
- Writing it into password manager. This is also bad, because in case of my inability/death/...., the share-hoders (eg. hairs) have no option how to recover funds. I can be also forced to open my password manager at some situations and having TREZOR physically and forcin the PIN -> all funds may be stolen.

HeroHann, 2021-02-05:
I also would like this. Any "product decision" already taken?

HeroHann, 2021-06-21:
I see this feature was added to the backlog. In my opinion it should be implemented in a timely manner.
Trezor advertises the "Phasphrase Feature" as the holy grail of security as it is the only 100% reliable way to be secure without a secure element.

That being said a lot of people use it and may not fully understand the importance of remembering that passphrase.
A Shamir Backup of it would help so many people (but it weakens the security I guess?).

There should be development on how to do it securely.

stepmes, 2021-08-17:
Thanks, @HeroHann, for your initiative, your suggestions here are appreciated, including this one, which would be a great addition to the advanced security options. We'll include it in our list of new feature ideas and will consider it in the future.

stepmes, 2021-08-17:
Please keep an eye on [our roadmap](https://github.com/orgs/trezor/projects/28). We update it with our priorities for the upcoming months and note their estimated dates of implementation.

closed 2021-08-17 by stepmes