trezor/trezor-firmware #1411: Trezor Firmware should ask for PIN before sending transactions

jingxi285, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
Why not ask for the PIN, or if no PIN, the passphrase before allowing transactions? It's very risky
Coinbase and Binance always ask me for 2FA before a payment can be processed.
Anyone could walk up to a laptop with an open Trezor Suite and make a quick payment, or do it via remote desktop attack etc

**Describe the solution you'd like**
Ask for PIN before putting through transactions

**Describe alternatives you've considered**
Could also ask for the passphrase again.

**Additional context**

I checked in Trezor-Suite-20.12.1-linux-x86_64.AppImage

Teralodom, 2021-01-09:
I don't see how this could help at all. 10 minutes is a reasonable time, you can decrease it to the minimum if that helps you.

Do you leave your PC with Trezor suite open, the device connected and unlocked for more than 10 minutes? I only connect my device when I need to do a transaction, and store it after that.

The exchanges use case is totally different, anyone can access your account at any time so they have to be more careful.

As a user, I would find it highly annoying to have something like this. But I guess it is ok if there was a way to activate it, definitively not default.

jingxi285, 2021-01-10:
> I don't see how this could help at all. 10 minutes is a reasonable time, you can decrease it to the minimum if that helps you.
>
> Do you leave your PC with Trezor suite open, the device connected and [...]

Thank you for your reply.

Often go for a cup of tea away from computer.

My bank asks for 2FA before making a payment.

Yes, I'd be happy if it could just be enabled on the device overall

stepmes, 2021-12-12:
Thanks @jingxi285  for this input, we're noting it down and will consider it when working on future improvements.

closed 2021-12-12 by stepmes