trezor/trezor-firmware #1419: Invalid seeds can be both restored and checked as valid, even with Invalid BIP39 checksum

3rdIteration, 2021-01-16:
**Describe the bug**
Both Trezor One and Trezor T

**Firmware version and revision**
Trezor One, firmware 1.9.3, Trezor T, firmware 2.3.4

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite Beta & Trezor Web Wallet
 - OS: Windows

**To Reproduce**
Steps to reproduce the behavior:
1. Wipe device.
2. Restore existing seed (eg: zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra zebra)
3. Seed should be rejected as invalid
4. Seed is accepted and device is initialised using it, all subsequent addresses are correctly derived from it. (The invalid seed)

Likewise recovery check also produces the same behavior. (Says seed is vaild, and if initialised with an invalid seed, will also say it matches what is on the device)

**Expected behavior**
Seed should be rejected as invalid...

**Screenshots**

**Additional context**

prusnak, 2021-01-16:
Duplicate of https://github.com/trezor/trezor-suite/issues/3262

closed 2021-01-16 by prusnak

3rdIteration, 2021-01-16:
Issue also occurs in Trezor web wallet

prusnak, 2021-01-16:
> Issue also occurs in Trezor web wallet

That's possible, but we are not introducing any new changes to the old wallet