trezor/trezor-firmware #1424: Click tests failing on actual hardware

mmilata, 2021-01-22:
Might be worth looking into this once #946 is fixed.
```
$ pytest -v tests/click_tests
===================================================================== test session starts =====================================================================
platform linux -- Python 3.8.6, pytest-6.1.1, py-1.9.0, pluggy-0.13.1 -- /home/mmilata/work/trezor-firmware/.venv/bin/python
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('/home/mmilata/work/trezor-firmware/.hypothesis/examples')
Using --random-order-bucket=module
Using --random-order-seed=776839

rootdir: /home/mmilata/work/trezor-firmware, configfile: setup.cfg
plugins: ordering-0.6, timeout-1.4.2, hypothesis-5.37.1, flaky-3.7.0, random-order-1.0.4
collected 11 items                                                                                                                                            

tests/click_tests/test_lock.py::test_hold_to_lock FAILED                                                                                                [  9%]
tests/click_tests/test_reset_slip39_advanced.py::test_reset_slip39_advanced_16of16groups_16of16shares FAILED                                            [ 18%]
tests/click_tests/test_reset_slip39_advanced.py::test_reset_slip39_advanced_2of2groups_2of2shares FAILED                                                [ 27%]Fatal Python error: Aborted

Current thread 0x00007f73c0b4e640 (most recent call first):
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 1318 in releaseInterface
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/transport/webusb.py", line 61 in close
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/transport/protocol.py", line 91 in end_session
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/transport/protocol.py", line 120 in end_session
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/client.py", line 107 in close
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/debuglink.py", line 384 in close
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/tools.py", line 233 in wrapped_f
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/device.py", line 190 in reset
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/tools.py", line 231 in wrapped_f
  File "/home/mmilata/work/trezor-firmware/python/src/trezorlib/tools.py", line 210 in wrapped_f
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/concurrent/futures/thread.py", line 57 in run
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/concurrent/futures/thread.py", line 80 in _worker
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/threading.py", line 870 in run
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f73d12062c0 (most recent call first):
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/threading.py", line 302 in wait
  File "/nix/store/m7080pw0ryjk0jhljp55rq1hd2qy8gki-python3-3.8.6/lib/python3.8/concurrent/futures/_base.py", line 434 in result
  File "/home/mmilata/work/trezor-firmware/tests/device_handler.py", line 51 in kill_task
  File "/home/mmilata/work/trezor-firmware/tests/device_handler.py", line 81 in check_finalize
  File "/home/mmilata/work/trezor-firmware/tests/conftest.py", line 294 in device_handler
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/fixtures.py", line 926 in _teardown_yield_fixture
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/fixtures.py", line 1012 in finish
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 394 in _callfinalizers
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 404 in _teardown_with_finalization
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 386 in _pop_and_teardown
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 426 in _teardown_towards
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 418 in teardown_exact
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 176 in pytest_runtest_teardown
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 84 in <lambda>
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 256 in <lambda>
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 310 in from_call
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 255 in call_runtest_hook
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/flaky/flaky_pytest_plugin.py", line 138 in call_and_report
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 128 in runtestprotocol
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/runner.py", line 110 in pytest_runtest_protocol
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/flaky/flaky_pytest_plugin.py", line 94 in pytest_runtest_protocol
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 84 in <lambda>
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/main.py", line 338 in pytest_runtestloop
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 84 in <lambda>
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/main.py", line 313 in _main
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/main.py", line 257 in wrap_session
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/main.py", line 306 in pytest_cmdline_main
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 84 in <lambda>
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/config/__init__.py", line 164 in main
  File "/home/mmilata/work/trezor-firmware/.venv/lib/python3.8/site-packages/_pytest/config/__init__.py", line 187 in console_main
  File "/home/mmilata/work/trezor-firmware/.venv/bin/pytest", line 8 in <module>
Aborted (core dumped)

```

tsusanka, 2021-10-05:
@mmilata is this issue still relevant?

mmilata, 2021-10-05:
Yes this is still failing even outside CI.