trezor/trezor-firmware #1427: Bootloader exceeds its size?

tsusanka, 2021-01-22:
I am actually having another failure running `./build-docker.sh` (cc @prusnak)

This time concerning legacy's bootloader:

```
./firmware_align.py bootloader.bin
Traceback (most recent call last):
  File "./firmware_align.py", line 11, in <module>
    raise Exception(
Exception: bootloader has to be smaller than 32736 bytes (current size is 32932)
make: *** [Makefile:26: align] Error 1
make: Leaving directory '/tmp/trezor-firmware/legacy/bootloader'
```

but we haven't touched the bootloader in quite some time have we? I could find only https://github.com/trezor/trezor-firmware/commit/b9c0b3969b03943eb6d23383208411a08fd17c17, could the one extra `if` statement be it?

prusnak, 2021-01-22:
That + we changed the GCC from GCC 9.x to GCC 10.x

I'll have a look.

prusnak, 2021-01-22:
Since we don't need to build the new bootloader in this release, quick fix would be to remove the following line:

https://github.com/trezor/trezor-firmware/blob/250f09245b657e29ae339f748b8ff2b3902dc350/legacy/script/cibuild#L19

prusnak, 2021-01-22:
> Since we don't need to build the new bootloader in this release, quick fix would be to remove the following line:

Applied this change to the release branch in 013f36981d2f9661c0853e10fe367702a7969466

I have some ideas what to do with the bootloader to reduce the size, but these are rather big changes and I don't want to do them right now postponing the freeze.

closed 2021-03-03 by prusnak