trezor/trezor-firmware #1430: Payment requests

tsusanka, 2021-01-25:
Tech Spec: [Google Doc](https://docs.google.com/document/d/1gnN2faUZ9Ymc2-QeKD5EcY6aBEOnyzx4h3fhD_N2H3o/edit#heading=h.a30oxabks97l)

andrewkozlik, 2021-01-25:
@matejzak regarding the design of showing output addresses: Note that there could be multiple outputs in the payment request, but the mainstream use-case is just one output. I imagine it could look something like this:
```
Confirm sending
0.05 TEST to
trezor.io

Buying 15.9636 DASH.

[grey "Details"] [green check-mark]
```

If there is just one output, then instead of "Address #1" we would show just "Address". Pressing "Details" would show:
```
Address #1
0.02 TEST to
tb1qkvwu9g3k2pdxew
fqr7syz89r3gj557l3
uuf9r9

[red cancel-mark] [grey "Next"]
```

Pressing "Next" would show:
```
Address #2
0.03 TEST to
tb1qldlynaqp0hy4zc
2aag3pkenzvxy65sae
sxw3wd

[red cancel-mark] [grey "Done"]
```

Pressing "Done" would show the initial "Confirm sending" screen again, where the user could confirm.

andrewkozlik, 2021-01-29:
I have been giving some more thought to the implementation and what I proposed above might be unnecessarily complicated, because it allows the user to repeatedly cycle through the outputs. Currently we don't even allow the user to go back and forth with normal outputs, so this might not be worth the effort at the moment. The simplest thing I can think of is allowing the user to just "unpack" the outputs for one pass-through:

```
Confirm sending
0.05 TEST to
trezor.io

Buying 15.9636 DASH.

[grey "Details"] [green check-mark]
```

Pressing "Details" would show:
```
Confirm sending
0.02 TEST to
tb1qkvwu9g3k2pdxew
fqr7syz89r3gj557l3
uuf9r9

[red cancel-mark] [green check-mark]
```

Pressing the cancel-mark would cancel the entire signing flow. Pressing the check-mark would show the next output in the payment request:
```
Confirm sending
0.03 TEST to
tb1qldlynaqp0hy4zc
2aag3pkenzvxy65sae
sxw3wd

[red cancel-mark] [green check-mark]
```

Pressing the check-mark would continue with the other outputs in the transaction. In other words, if the user presses [Details], then the outputs that are packed inside the payment request would be shown just as if there was no payment request, otherwise they would be hidden.

tsusanka, 2022-01-24:
Note from Slush: it would be great if we could incorporate "refund address" into payment requests. @andrewkozlik do you think you could discuss it with him on some occasion @andrewkozlik?

prusnak, 2022-01-24:
> Note from Slush: it would be great if we could incorporate "refund address" into payment requests. @andrewkozlik do you think you could discuss it with him on some occasion?

I'll discuss this with Andrew in context of https://github.com/trezor/trezor-suite/issues/4800

closed 2022-01-31 by andrewkozlik