trezor/trezor-firmware #1441: Enforce changelog entries via CI

matejcik, 2021-02-01:
something along the lines of:
if there's a change in X subdir, AND the change is not isolated to `messages` (i.e. regenerated protobuf), there MUST be a change in the `X/CHANGELOG.md` file, otherwise the task fails

This would check that we add proper changelogs, and we could decide to merge a PR even if a changelog is missing, in case that is appropriate.

There is an option to set this CI task to "allowed to fail", but that would mean that the CI pipeline would show green even with missing changelogs, right?

tsusanka, 2021-02-04:
I would suggest adding this as a github action. Some options:

- https://github.com/marketplace/actions/changelog-checker
- https://github.com/marketplace/actions/changelog-enforcer
- https://github.com/marketplace/actions/changelog-reminder

The last is very simple: it only checks if CHANGELOG was touched and if not it comments the PR with "you are missing a changelog entry". And you can decide to ignore it and carry on. Isn't that enough?

The first two are a bit smarter and AFAICT they indeed fail as a CI if the changelog is missing. If you indeed do not want to add changelog entry you add a label such `no changelog` and that's it. 

Any preference?

tsusanka, 2021-02-04:
Example of the changelog-reminder: https://github.com/tsusanka/trezor-firmware/pull/1

matejcik, 2021-02-05:
i think the changelog checker is a good start. How exactly does it work though? Will it register change in _any_ changelog, or does it need _all_ changelogs?

tsusanka, 2021-02-12:
Hm I am worried the aforementioned tools do not support multiple Changelogs :-/.

tsusanka, 2021-02-15:
@mmilata will try out https://github.com/twisted/towncrier

mmilata, 2021-02-15:
Played with towncrier a bit and it seems usable (see twisted/towncrier#128 for more details wrt markdown):

* see 5eaddde57946567a49fe07f8a184a050038e2ece for example configuation and couple of changelog entries
* see 610ffbfac7ede43cd6ad37fa1a1f14315bb75dab for resulting CHANGELOG.md ([rendered](https://github.com/trezor/trezor-firmware/blob/610ffbfac7ede43cd6ad37fa1a1f14315bb75dab/core/CHANGELOG.md))
* It requires each entry to have github issue. For entries without issue it is recommended to create PR first, then use that PR number.

To use it we'd need to:
* configure it for all 4 changelogs we have
* make it use our date format
* write documentation for developers
* add CI task that enforces creation of changelog
* change `make changelog` to run `towncrier build --draft`
* hook `towncrier build` into the release process (it needs version number passed to it), use `linkify-changelogs.py` on generated file

PR #1575 (2021-05-04): Use towncrier for generating changelogs
Fixes #1441. Usage: [docs/misc/changelog.md](https://github.com/trezor/trezor-firmware/blob/mmilata/towncrier/docs/misc/changelog.md)

Notes:
* [Towncrier docs](https://github.com/twisted/towncrier)
* `make changelog` was removed, `make changelog_check` now runs `towncrier build --draft` on all subprojects which is probably not that useful so maybe I should remove it too.
* The CI job has `master` hardocoded as the base branch, AFAIK there's no way to for us to obtain it in gitlab CI?
* The `tools/generate-changelog.py` stages changes to git. This is tow [... cut]
closed 2021-05-04 by mmilata