trezor/trezor-firmware #1451: Trezor model T Binance smart chain: "unsupported chain id" - can't sign transactions

TimStone84, 2021-02-06:
**Describe the bug**
Very similar issue to: https://github.com/trezor/trezor-firmware/issues/1327

Upgraded to Trezor Model T and imported Trezor 1 seed. I can sign transactions on the ETH network on newest firmware but getting "transaction failed unsupported chain ID" on Binance smart chain.
Getting this error on 2.3.3 and 2.3.4

Downgraded to 2.3.0 which solves it and I can sign transactions with no problems

Trezor 1 works as intended on newest firmware with no issues, problem is only with newer firmwares ModelT

Is this a known issue and will this be fixed in an upcoming firmware?

Should I rather continue using my Trezor 1 on newest firmware or the model T on 2.3.0 until the issue is fixed?

thanks alot in advance

prusnak, 2021-02-06:
@matejcik this should be fixed in the upcoming FW 2.3.5, right?

matejcik, 2021-02-08:
that is correct.

closed 2021-02-08 by matejcik

hfray, 2021-02-20:
> **Describe the bug**
> Very similar issue to: #1327
> 
> Upgraded to Trezor Model T and imported Trezor 1 seed.
> Using Trezor via metamask.
> 
> I can sign transactions on the ETH network on newest firmware but getting "transaction failed unsupported chain ID" on Binance smart chain.
> Getting this error on 2.3.3 and 2.3.4
> 
> Downgraded to 2.3.0 which solves it and I can sign transactions with no problems
> 
> Trezor 1 works as intended on newest firmware with no issues, problem is only with newer firmwares ModelT
> 
> Is this a known issue and will this be fixed in an upcoming firmware?
> 
> Should I rather continue using my Trezor 1 on newest firmware or the model T on 2.3.0 until the issue is fixed?
> 
> thanks alot in advance

how did you downgrad your firmware. IM having this issue right now

TimStone84, 2021-02-20:
> how did you downgrad your firmware. IM having this issue right now

it should work if you are on the newer versions. at least it does for me. they solved it with 2.3.5

alternatively you just download the old firmware and flash it on your drive. google is your friend