trezor/trezor-firmware #1453: Verifying Stellar Wallet Address is cut off on Trezor One Device Screen

dereileak, 2021-02-06:
While using Exodus to receive Stellar coins on the Trezor One, viewing the wallet address on the device to receive causes the end of the address to be cut off while verifying

Trezor Firmware:  1.9.3.

 - Wallet software: Exodus
 - OS: Mac Catalina 10.15.7
 - Exodus Version: 21.1.29 

Reproduce:
1) Connect wallet and put in passphrase, testing with no passphrase
2) Click wallet link button to view contents
3) Click Stellar coins to receive
4) click receive then review on device
5) Address will be presented in the app
[image]
6) Device will show cut off Address
[image]

Expected behavior:
The full address should be viewable on the trezor device

bosomt, 2021-12-28:
I was able to reproduce issue in Exodus 21.12.17 and in https://accountviewer.stellar.org/dashboard
Last characters of address are missing on Trezor One 1.10.2

sime, 2023-01-23:
Similar issue on MATIC. ETH is also reported.

[image]

Hannsek, 2023-02-20:
related?: https://github.com/trezor/trezor-firmware/issues/1212

closed 2023-02-27 by matejcik

bosomt, 2023-02-27:
**QA NOK**

```
 ~/Downloads  trezorctl stellar get-address                                                                                                                           
GAXSFOOGF4ELO5HT5PTN23T5XE6D5QWL3YBHSVQ2HWOFEJNYYMRJENBV
```

but on display i see `GAXSFOOGF4ELO5HT5PTN23T5XE6D5QWL3YBHSVQ2HWOFEJNYYMRJE` im missing 3 last characters ;(

 - Device: model 1 1.11.3 Universal (revision 5e8f3238de9c9527648950d93c9c444a74d043ef)

matejcik, 2023-02-27:
that could be because this will be fixed in the upcoming release

bosomt, 2023-02-27:
**QA OK**

sorry wrong branch ;)

 - Device: model 1 1.11.3 Universal (revision be920d7cec2495d82496af500cbdeec6b945c505)

commit cd91bcd0 (2023-02-27): fix(legacy): use correct layout for StellarGetAddress (fixes #1453)
commit bf711d33 (2023-02-27): fix(legacy): use correct layout for StellarGetAddress (fixes #1453)
PR #2848 (2023-02-27): T1 layout fixes
fixes #2373
fixes #1453 
two bugs that look very similar but in fact are in completely different code paths

also fixes a small bug in UI reports, and I added Ethereum address screens to the UI fixtures
commit c40f8a6a (2023-02-27): fix(legacy): use correct layout for StellarGetAddress (fixes #1453)