trezor/trezor-firmware #1455: U2F is incompatible between One and T

fkirc, 2021-02-07:
I run into this issue again with newer firmware: https://github.com/trezor/trezor-core/issues/276
Unfortunately, this means that I could potentially lockout myself if my Model One is the only backup device for my Model T (both using the same seed phrase).

**Firmware version and revision**

1.9.3

**To Reproduce**

- Configure a Model T and a Model One with the exact same recovery seed
- Go to https://www.dropbox.com/account/security
- Add the Model T as a "Security Key"
- Logout ouf Dropbox
- Try to login again with the Model One -> Get error message on Model One screen: `another device was used to register...`

**Expected behavior**

Logout and Login should work seamlessly with two different Trezors that use the same seed phrase.

prusnak, 2021-02-07:
This is different issue than https://github.com/trezor/trezor-core/issues/276

Trezor Model T uses FIDO2
Trezor One uses U2F (FIDO1)
Dropbox picks FIDO2 if possible and only fallbacks to FIDO1 if not available

If you want to recovery the FIDO2 behaviour, you need to have FIDO2 capable device (that is Model T).

closed 2021-02-07 by prusnak

fkirc, 2021-02-07:
> This is different issue than [trezor/trezor-core#276](https://github.com/trezor/trezor-core/issues/276)
> 
> Trezor Model T uses FIDO2
> Trezor One uses U2F (FIDO1)
> Dropbox picks FIDO2 if possible and only fallbacks to FIDO1 if not available
> 
> If you want to recovery the FIDO2 behaviour, you need to have FIDO2 capable device (that is Model T).

Thanks for the quick response.
I just tried this with Google, and it seems that Google works seamlessly with both devices.
So am I right with the assumption that Google uses FIDO1?

prusnak, 2021-02-07:
> So am I right with the assumption that Google uses FIDO1?

Exactly. Google does not support FIDO2 yet.