trezor/trezor-firmware #1460: Fix size checks in msg_read_tiny

andrewkozlik, 2021-02-09:
I think that there are some bugs in `msg_read_tiny()` having to do with d6f41dba9e. This commit increased the size of `msg_tiny`, but didn't update the related code.
1. There is a magic number 64 in `msg_read_tiny()`, `main_rx_callback()`, `debug_rx_callback()`, `usbPoll()`, which probably matches the old size of the `msg_tiny` buffer but which probably wasn't updated.
2. There are three checks at the beginning of the function which return, but don't generate an error message. 

As a result of these two issues, when I increased the size of the `pin` field in `PinMatrixAck` it passed the static checks, but failed at runtime due to point 1 and did not return any error code due to point 2.

prusnak, 2021-02-09:
@jhoenicke do you remember why we increased `msg_tiny` size from 64 bytes to 128 bytes?

jhoenicke, 2021-02-09:
> @jhoenicke do you remember why we increased msg_tiny size from 64 bytes to 128 bytes?

This was when state was added to the `PassphraseAck` message.

invd, 2021-02-09:
This change was put in place to avoid out of bounds issues, as discovered by my Trezor fuzzer in early 2018.
I actually have some writeup on this topic through the issue reappearing in a fork, see [this blog post](https://blog.inhq.net/posts/skycoin-firmware-vuln/#details-for-vulnerability-no-1).

invd, 2021-02-09:
@andrewkozlik I haven't had time to go over the specifics, AFAIK the `64` in many of those places is coming from the fact that the raw USB packets should always be exactly 64 bytes in size, which is (mostly) unrelated to the actual size of the `msg_tiny[128]` buffer.

andrewkozlik, 2021-02-09:
> > @jhoenicke do you remember why we increased msg_tiny size from 64 bytes to 128 bytes?
> 
> This was when state was added to the `PassphraseAck` message.

That's funny, because one thing that I tried was reducing the `tiny_msg` size back to 64 bytes to see what which of the messages would trigger the static asserts. It turns out everything worked! It seems the reason might be that the `state` field has been deprecated.

@invd thanks for explaining the magic 64, we should definitely add a constant to clarify this.

andrewkozlik, 2021-02-09:
PR https://github.com/trezor/trezor-firmware/pull/1466

andrewkozlik, 2021-02-09:
One more thing that puzzles me. During the build we generate `messages_map_limits.h` in `legacy/firmware/protob`. It has all these size checks, like:
```
_Static_assert(MSG_IN_SIZE >= sizeof(SignTx), "msg buffer too small");
```
but it seems to me that it should actually be checking something like
```
_Static_assert(MSG_IN_SIZE >= SignTx_size, "msg buffer too small");
```
because `SignTx_size` is the size of the protobuf encoded message whereas `sizeof(SignTx)` is the size of the decoded message in memory.

prusnak, 2021-02-09:
> because `SignTx_size` is the size of the protobuf encoded message whereas `sizeof(SignTx)` is the size of the decoded message in memory.

We use `MSG_IN_SIZE` in two different contexts:

https://github.com/trezor/trezor-firmware/blob/32c37aa9cdc982068c27b2ee9b48ee225cc21232/legacy/firmware/messages.c#L217

https://github.com/trezor/trezor-firmware/blob/32c37aa9cdc982068c27b2ee9b48ee225cc21232/legacy/firmware/messages.c#L230

* `msg_data` is decoded message
* `msg_in` is raw message

It would be probable better if we split `MSG_IN_SIZE` into `MSG_RAW_IN_SIZE` and `MSG_DECODED_IN_SIZE` and then using the following two checks:

```
_Static_assert(MSG_DECODED_IN_SIZE >= sizeof(SignTx), "msg buffer too small");
_Static_assert(MSG_RAW_IN_SIZE >= SignTx_size, "msg buffer too small");
```

closed 2021-03-17 by andrewkozlik