trezor/trezor-firmware #1462: Transaction passed but I never holded the confirm button

bitcoin4cashqc, 2021-02-09:
**Describe the bug**
I was going to sell ETH on uniswap for stable coins. I decided to NOT confirm the tx by NOT holding the hold green button. I bought like 2 eth worth of a shitcoin instead. I woke up this morning and the sell eth tx was passed, not the shitcoin.

**Firmware version and revision**
Latest firmware on model T

**Desktop/smartphone setup (please complete the following information):**
 - trezor suite on android chrome.

**To Reproduce**
Steps to reproduce the behavior:
1. Went to uniswap, initiated a sell eth tx but canceled at the hold green button step.
2. Switched to the shitcoin and changed the buy token side for the shitcoin
3. Confirmed this shitcoin tx
4. Wokeup in a loss

**Expected behavior**
How the hell is that would be possible. Whats the point kf asking an hold confirm..

closed 2021-02-09 by prusnak

bitcoin4cashqc, 2021-02-10:
No mate.

I'm sorry I said I was one mobile. I was not. Was on metamask and ubuntu.

So, I went for confirming the tx on my trezor (by hoding the confirm hold green button BUT I clearly not confirmed.) 

My friend called me and I decided to wait before selling eth. I decided to CANCEL the tx by NOT holding confirm button but clicking the X.

I then made a smaller trade for eth->shitcoin.

What I think happenned is for some reasons, the tx data/bytecode of the sell eth for stable was kept in the trezor/metamask/chrome cache and I confirmed this one when I was actually expecting to confirm the sell smaller amount of eth for shitcoin.

Is that possible?

It have nothing to do with uniswap itself and their support 100% sure about this.

matejcik, 2021-02-11:
@bitcoin4cashqc the important part is, what was on **Trezor** screen when you confirmed?

presumably a screen like [this](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/1022973762/artifacts/test_ui_report/passed/test_msg_ethereum_signtx_eip155.py::test_chain_ids[1-60-sig0].html) ?
[image]
[image]

Did you check the destination address and the amount?

bitcoin4cashqc, 2021-02-12:
Yes and I didnt checked. But if I hitted X instead of hold to confirm, then move on to the other tx, why and where the old would have been stored?

Téléchargez Outlook pour Android<https://aka.ms/ghei36>

[image]<https://user-images.githubusercontent.com/596073/107616753-f5adf780-6c4e-11eb-87c9-d23273ff4bcf.png>
[image]<https://user-images.githubusercontent.com/596073/107616775-fcd50580-6c4e-11eb-9e9e-d1352e62c6e9.png>

matejcik, 2021-02-15:
> where the old would have been stored?

in metamask presumably? if you cancelled your transaction, a new one must have been sent from the host. If the host sent the same tx again, and you didn't check and confirm, then the same tx would happen.

> why

who knows