trezor/trezor-firmware #1467: Model T no longer allows signing for arbitrary deriviation paths below `m/45'`

dhruvbansal, 2021-02-10:
**Describe the bug**

Signing for arbitrary public keys with deriviation paths under `m/45'` used to work -- it no longer does.

**Firmware version and revision**
Model T 2.3.5

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Unchained Capital platform & Caravan
 - OS: Seen on macOS but believed to be OS-independent issue
 - Browser: Seen on Chrome 87, but believed to be browser-independent issue

**To Reproduce**
Steps to reproduce the behavior:
1. Construct a transaction using public keys derived from a path such as `m/45'/0'/0'/0/0/0`(6 path components which, indeed is different than the BIP45 spec).
2. Sign the transaction with a pre-2.3.5 model T or with a Trezor ONE -- it works.
3. Sign the same transaction with a 2.3.5 model T, we get `Forbidden key path` error.

**Expected behavior**
We expected to be able to continue to sign for arbitrary derivation paths below `m/45'`.  This changed without any notification in advance, breaking signing for many of our customers.

**Screenshots**
Below is a screenshot of Caravan's [functional test suite](https://unchained-capital.github.io/caravan/#/test)
[image]

Below are two screenshot of Unchained Capital's (test) application, first when signing for a transaction
[image]

And then when confirming an address.
[image]

prusnak, 2021-02-10:
Suggestion from Unchained Capital mentioned on Slack:

```
PATTERN_UNCHAINED_A = "m/45'/coin_type'/account'/[0-1000000]/change/address_index"
PATTERN_UNCHAINED_B = "m/45'/coin_type/account/[0-1000000]/change/address_index"
PATTERN_UNCHAINED_C = "m/45'/coin_type’/account’/[0-1000000]/address_index"
```

@dhruvbansal range `[0-1000000]` seems a little bit excessive - are you really using that wide range?

prusnak, 2021-02-13:
Fixed via https://github.com/trezor/trezor-firmware/pull/1473

closed 2021-02-13 by prusnak