trezor/trezor-firmware #1469: PIN entry correction on TT

andrewkozlik, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
The PIN entry screen on the TT shows dots instead of the entered digits. If the user is unsure whether they made a mistake while entering the PIN, then they have no way to check and have to either re-enter the entire PIN again or risk losing a PIN entry attempt. Even if the user knows that they made a mistake, pressing the "backspace" button erases the entire PIN, so they have to retype the whole thing. 

**Describe the solution you'd like**
- Allow the user to temporarily reveal the PIN that they are entering by touching the entry field. As soon as they remove their finger the field will show dots again.
- Change the behavior of the backspace button so that it erases one digit at a time. If pressed and held for 2 seconds or more, then make it erase the whole field.

**Additional context**
This is related to the implementation of the unlimited length PIN https://github.com/trezor/trezor-firmware/issues/1167. Long PINs make mistakes not only more likely but also more costly if the user doesn't have a way to correct them. Note that the entry field is currently center-aligned. For long PINs we want to make it behave like the passphrase entry field.

mmilata, 2022-04-21:
Opened #2227 which should address this in the new design. Feedback appreciated, firmware image for testing: [firmware.bin.gz](https://github.com/trezor/trezor-firmware/files/8534873/firmware.bin.gz)

```
trezorctl device setup --skip-backup  # --skip-backup required as backup not implemented yet
trezorctl set pin
trezorctl clear-session
trezorctl ping ping  # or tap on the screen
```

sime, 2022-05-04:
@mmilata Tested ACK

Happy path works like a charm.

andrewkozlik, 2022-05-04:
Tested. Works great!

closed 2022-05-16 by mmilata

mmilata, 2022-05-16:
QA: as the fix affects new UI only it is not possible to test in release firmwares, moving it to approved.

@hynek-jina do we track things that need to be tested in new UI somewhere?

bosomt, 2022-12-13:
**QA OK** :) 

**Info:**
 - Suite version: desktop 22.12.1 (ddaacdb2245d7a909b3d2abfe918cff34d2e1609)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.12.1 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ff35c77bad82c7acac1df6245940ebe0d6454642)
 - Transport: bridge 2.0.32

PR #2227 (2022-05-16): Rust UI: PIN correction + long PIN
Fixes #1469.

[image]

* yellow button now erases one character and holding it for 2s erases everything
* holding finger on dot display reveals the pin
* PIN can be up to 50 digits long and renders reasonably even if it doesn't fit screen
* overflowing PIN animated (yet) but using the same "twitching" as in original UI