trezor/trezor-firmware #1471: Trezor T with 3rd party wallet Qtum-Electrum - unknown path m/44'/88’/0’/0/0

SereFF1984, 2021-02-11:
Hello! 
Trezor Model T, firmware 2.3.5
Trezor One, firmware 1.9.4
Qtum-electrum-v4.0.11

Trezor support Qtum via Qtum-Electrum.
[image]
In March 2020 was created new wallet in Qtum-electrum 0.18.16 with Trezor.
Selected legacy(p2pkh) with default path: m/44’/88’/0’
[image]
The wallet was created successfully and I deposit some Qtum coins to it. The coins were deposited successfully.
[image]
[image]
Today I tried to send coins from this wallet but I got an unknown path m/44’/88’/0’/0/0 in Trezor T
[image]
and error in Qtum-Electrum. So coins didn't send.
[image]
Tried to do something in Trezor One (firmware 1.9.4) but got the following errors:
[image]
[image]
and error in Qtum-Electrum:
[image]
Is there a solution to solve this problem with wrong address part in Trezor?

prusnak, 2021-02-11:
According to SLIP-0044: https://github.com/satoshilabs/slips/blob/master/slip-0044.md

the correct value for QTUM is 2301, not 88

Developers of Qtum fixed that in October 2019 in Qtum-Electrum 4.0.0 - see here https://github.com/qtumproject/qtum-electrum/commit/3e9cd1d78057681da6a0e2c466501580b21cabab

So I am not sure how you managed to create these addresses on wrong paths, since the issue should have been fixed already.

To get around the limitation you can switch the Trezor safety checks to prompt via:

`trezorctl set safety-checks prompt`

See https://wiki.trezor.io/Using_trezorctl_commands_with_Trezor#Safety-checks for more info

closed 2021-02-11 by prusnak

icodeface, 2021-03-02:
Hi @prusnak, `trezorctl set safety-checks prompt` seems not work on Trezor One using firmware 1.9.4 , I got `Error: DataError: No setting provided`.

prusnak, 2021-03-02:
@icodeface This works only with Trezor Model T. The setting can't be applied for Trezor One.

icodeface, 2021-03-02:
> @icodeface This works only with Trezor Model T. The setting can't be applied for Trezor One. With Trezor One you have three options:
> 
> * try downgrading to firmware 1.9.0 (`trezorctl firmware-update -v 1.9.0`)
> * get Trezor Model T and import the seed
> * import the seed into Electrum (not recommended as this exposes all your coins to the computer which might be running malware)

Thanks for replying, is there any possibility add the safety-checks level function to Trezor One in the next firmware release ?

prusnak, 2021-03-02:
> Thanks for replying, is there any possibility add the safety-checks level function to Trezor One in the next firmware release ?

No