trezor/trezor-firmware #1474: When deploying a smart contract from Trezor through MetaMask, forbidden key path error is received

ravisriv, 2021-02-14:
**Describe the bug**
In firmware version 2.35 (Trezor Model T), it is not possible to deploy or interact with an Ethereum smart contract through a wallet stored in Trezor connected through the hardware wallet connection option in MetaMask. The user will repeatedly get the error: Forbidden key path

**Firmware version and revision**
This error occurs in Trezor Model T firmware version 2.35
When the firmware is downgraded to version 2.30, the error is fixed. Therefore, the error appears to have been introduced after version 2.30. I have found this issue on version 2.35. I have not tested other versions that came after 2.30.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Model T firmware version 2.35 through the hardware connection option in MetaMask
 - OS: Windows 10 Pro
 - Browser Chrome
 - Version 88.0.4324.150

**To Reproduce**
Steps to reproduce the behavior:
1. Setup an Ethereum wallet on Trezor Model T (with firmware version 2.35)
2. Connect to Trezor wallet using MetaMask Chrome browser extension (Hardware wallet connection options)
3. Try to deploy a simple smart contract using the Ethereum wallet stored in Trezor
4. You'll get the error "forbidden key path."

**Expected behavior**
The behavior of the Trezor wallet (with firmware version 2.35) should be no different than the way it behaved in version 2.30. The hardware wallet should ask the user to confirm execution of the contract. Confirm payment of "gas". After that it should allow the smart contract to be deployed (when the gas is paid).

**Screenshots**
Sorry, cannot share screen.

**Additional context**
Please review the following link. It appears that others have noticed the same problem.
https://moneyonchain.com/blog/getting-an-error-using-trezor-maybe-you-should-downgrade-your-firmware/

prusnak, 2021-02-14:
What path does the token/smart contract use? I am pretty sure it uses path that does not belong to the Ethereum. Can you list it here?

ravisriv, 2021-02-14:
Hi @prusnak - Thank you for your comment. I'm using the default Ethereum wallet. The derivation path is /m/44'/60'/0'/0/0

Everything works as expected in Model T firmware version 2.30 but when I upgrade Trezor to firmware version 2.35, I start getting this forbidden key path error, whenever I try to use this default Ethereum wallet through MetaMask.

ravisriv, 2021-02-14:
[image]

ravisriv, 2021-02-15:
Also, the issue appears, if I try to transfer ETH or any ERC20 token from my Ethereum wallet on Trezor Model T (firmware 2.35) to any other account after connecting through MetaMask. So, the issue does not seem to be limited to the execution of smart contracts. Again, this issue does not exist in firmware version 2.30.

matejcik, 2021-02-15:
@ravisriv are you using a real Ethereum network or a testnet?

Testnets are forbidden from using the real Ethereum keys at 44'/60'/*. It's a metamask problem that it is using the same derivation paths.

ravisriv, 2021-02-15:
Thanks @tsusanka - I was using Rinkeby testnet for this. Give me a couple of days and I'll test this same functionality on Ethereum Mainnet. I agree that it makes sense to forbid testnet from using the same path.

prusnak, 2021-02-15:
> I was using Rinkeby testnet for this.

Then you are hitting the bug in Metamask which is using `m/44'/60'` paths even for testnet while actually `m/44'/60'` should be used for (all) testnets.

prusnak, 2021-02-15:
Reported to Metamask here - https://github.com/MetaMask/metamask-extension/issues/10450

closed 2021-02-15 by prusnak

aloisklink, 2021-03-06:
For anyone else coming across this issue, (especially since it seems like it might take a while for MetaMask to fix this), you can get around this without rolling back your firmware by installing `trezorctl` and setting safety-checks to `prompt`.

I can confirm I can send transactions with a Trezor Model T v2.3.6 on the Ethereum Goerli testnet after running this command.

From https://wiki.trezor.io/Using_trezorctl_commands_with_Trezor#Safety-checks:

> Set to "prompt" if you want to be able to allow potentially unsafe actions, such as mismatching coin keys or extreme fees.
> **This is a power-user feature. Use with caution.**
>
> Example:
> ```bash
> trezorctl set safety-checks prompt
> ```