trezor/trezor-firmware #1489: Wrong slip44 for Energy Web Chain resulting in "Forbidden Key Path" error

Mrtenz, 2021-02-18:
**Describe the bug**
The `slip44` value for Energy Web Chain is listed as 60 (the chain ID of Ethereum), rather than 246, in the [`networks.py`](https://github.com/trezor/trezor-firmware/blob/bd0b91a229473c23c224f7dd0e3ee50e53480059/core/src/apps/ethereum/networks.py#L345) file. When trying to sign a transaction with a derivation path like `m/44'/246'/0'/0/0` on MyCrypto this results in Forbidden Key Path, even though this path should be allowed. It looks like the `verify_path` function [here](https://github.com/trezor/trezor-firmware/blob/7b9f3c046279b11752c60e5616b4643597d6b75b/core/src/apps/common/keychain.py#L118) throws the error.

In [`slip-0044.md`](https://github.com/satoshilabs/slips/blob/master/slip-0044.md), EWC is (correctly) listed as `246'`.

**Firmware version and revision**
Firmware version 2.3.6.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: MyCrypto web and desktop versions
 - OS: Arch Linux
 - Browser Google Chrome
 - Version 88.0.4324.182

**To Reproduce**
Steps to reproduce the behavior:
1. Connect through the MyCrypto desktop application
2. Select an EWC address (`m/44'/246'/...`)
3. Attempt to sign a transaction

**Expected behavior**
The Trezor device should not deny the transaction.

FrederikBolding, 2021-02-18:
This can also be reproduced directly in the Connect Explorer:
https://trezor.github.io/connect-explorer/#/method/ethereumSignTransaction

By signing
`{
    nonce: '0x0',
    gasPrice: '0x14',
    gasLimit: '0x14',
    to: '0xd0d6d6c5fe4a677d343cc433536bb717bae167dd',
    chainId: 246,
    value: '0x0',
    data: '0x'
}` using the following path `m/44'/246'/0'/0/0`

matejcik, 2021-02-18:
please fix this in the upstream `chains` repo. The EWC entry is missing the `slip44` entry, which means Trezor falls back on Ethereum
https://github.com/ethereum-lists/chains/blob/master/_data/chains/eip155-246.json

closed 2021-02-18 by matejcik

FrederikBolding, 2021-02-19:
Done @matejcik 
https://github.com/ethereum-lists/chains/pull/163