trezor/trezor-firmware #1490: Show error screen when Trezor runs out of memory

prusnak, 2021-02-18:
Currently when Trezor runs out of memory while decoding the protobuf message (during signing) it will throw a `MemoryError` exception and stops processing it. Host is not aware that something went wrong and so is not user because they'll still see the signing progressbar on the screen.

If we showed an error dialog on the device this would make the UX better.

tsusanka, 2021-02-19:
@matejcik @prusnak this should be quite straightforward right? If it is easy I would go for it.

matejcik, 2021-02-19:
_Some_ memory errors are already caught by the handler in `trezor.wire` and reported as `FirmwareError`s. This entails figuring out where the others are slipping through. But we do have tracebacks so this should not be too bad.

prusnak, 2021-02-19:
If we can report the `MemoryError` failure via `FirmwareError` to the host, we don't strictly need to show the error dialog on the device. But it would be better if we could.