trezor/trezor-firmware #1495: incorrect address / amount in Trezor One final confirmation screen

GordonGit, 2021-02-20:
Using Electrum with a Trezor One, the final  device confirmation screen on the Trezor device displays an a different address (a change address owned by the Electrum wallet) and an incorrect, higher amount of BTC to send. 

Firmware version 1.9.1

Electrum 4.0.2 / 4.0.9
Windows 10

Open Electrum
Enter Trezor PIN to unlock wallet
Go to Electrum "Send" tab, enter destination address and amount
Click "Pay"
Click "Finalize" if using advanced payment settings
Click "Sign"
Enter PIN
Confirm on 1st Trezor confirmation screen (details correct)
2nd confirmation screen displays different address and amount
Cancel transaction

I expected both confirmation screens to match, as per usual.

[image]

[image]

[image]

It seems the final confirm screen is showing an output to a change address owned by the wallet (compare red arrows in confirm screen 2 pic to electrum construction pic). This is unexpected and a bit disconcerting - I expect the tx would work out OK but I'm not comfortable risking it.

prusnak, 2021-02-20:
you can't have more than one change address - otherwise you need to confirm all change addresses - even if they belong to you

closed 2021-02-20 by prusnak

prusnak, 2021-02-22:
FTR this was caused by a setting in Electrum:

```
"Preferences">"Transactions">"Use multiple change addresses"
```