trezor/trezor-firmware #1498: trezorctl command to turn on Trezor to compute checksum word from manually entered incomplete seed

zdeneksvarc, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
When using diceware I can't compute checksum only with pen and paper. If I don't want to use computer, even offline, I can still use the trezor device. Why don't let the trezor device generate the checksum word from manually entered incomplete seed? 

**Describe the solution you'd like**
trezorctl command to turn on Trezor to compute BIP-39/SLIP-39 checksum word from manually entered incomplete seed.

**Describe alternatives you've considered**
Computing checksum word with software utility like [bip39check](https://github.com/bharathrao/bip39check ) in trusted offline environment like [Tails](https://tails.boum.org/).

**Additional context**
`trezorctl get-checksum -w [bip39|slip39] -t [128|192|256]`

matejcik, 2021-02-24:
basically a dup of #23

prusnak, 2021-02-24:
Closing as duplicate of https://github.com/trezor/trezor-firmware/issues/23

closed 2021-02-24 by prusnak