trezor/trezor-firmware #1501: Unable to Transfer Funds with Trezor Model T from a Metamask Account on Binance (BSC) Network

casoCornelius, 2021-02-25:
I have a latest version Trezor (20.12.1), Trezor Model T. I set up my Metamask account on the Binance Smart Chain (BSC) Network. When I attempt to withdraw funds from that account (Trezor-protected), I get "Tx Failed. Unknown Chain ID). I read online that someone who downgraded their Trezor firmware to 20.3.0, he could make it work that way. Can someone please assist me in downgrading my Trezor firmware from 20.12.1 to 20.3.0? Appreciations

prusnak, 2021-02-25:
Latest version of Firmware is 2.3.6. Please install that one - instructions - https://wiki.trezor.io/User_manual:Updating_the_Trezor_device_firmware

closed 2021-02-25 by prusnak

casoCornelius, 2021-02-25:
i will but do you think this is the error I'm causing?

casoCornelius, 2021-02-25:
that was really helpful - i've solved my issue. Thank you!

gipsonstella, 2021-04-02:
Has the error been solved?