trezor/trezor-firmware #1505: set-passphrase-source command not found in trezorctl

Th3Tr00p3r, 2021-03-01:
**Describe the bug**

Trying to set the passphrase source to 'device' using trezorctl CLI ([link](https://wiki.trezor.io/Using_trezorctl_commands_with_Trezor#Set_passphrase_source)) yields the following message:
`Error: No such command 'set-passphrase-source'.`

This is an attempt to work with the SimpleStaking wallet desktop app for Tezos, which does not handle passphrase input on host (a different issue, probably theirs - but since you [promote that wallet](https://wiki.trezor.io/Tezos_(XTZ))...)

I have used other commands and did the initial setup (including firmware update) with trezorctl.

**Firmware version and revision**
Trezor T, firmware version 2.3, trezorctl, version 0.12.2

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor CLI
 - OS: Windows 10

**To Reproduce**
Steps to reproduce the behavior:
input `trezorctl set-passphrase-source device` in command line

**Expected behavior**
a confirmation message appears, probably having to confirm on the Trezor device. the device now enforces passphrase typing on itself.

prusnak, 2021-03-01:
@matejcik what's the current state of things wrt the passphrase source?

matejcik, 2021-03-01:
does not exist anymore. passphrase source is driven by host.

in order to force on host, you instead use `trezorctl set passphrase enabled --force-on-device`

the wiki page for trezorctl is seriously out of date and we should probably do something about it. at the very least, put up a marker at top that it's out of date

prusnak, 2021-03-01:
Thanks!

@Th3Tr00p3r use `trezorctl set passphrase enabled --force-on-device` to force entering passphrase on the device

closed 2021-03-01 by prusnak

Th3Tr00p3r, 2021-03-01:
Thanks!
(It didn't solve my issue with SimpleStaking Wallet btw, I'll try opening an issue there if there isn't one)