trezor/trezor-firmware #1516: Update to Stacks v2

raymonddurk, 2021-03-05:
Stacks (Formally Blockstack) was supported for the v1 version of the protocol. As I currently understand it, the Model T will no longer work as they have forked into v2. I know the goal is to only support top 30 coins which Stacks is currently not one of them. I wanted to get a sense of any plans to update to v2?

raymonddurk, 2021-03-05:
Side note, but this very well may be on them given their blog post here.
https://www.hiro.so/questions/how-can-i-use-my-trezor-device-with-the-stacks-wallet

prusnak, 2021-03-05:
We are not going to do support his unless the coin developers send a pull request with the required changes. For t's not feasible to devote our development power to support all coins hardforks. And if coin developers can't spend their resources on a hardware wallet support, it's their choice.

closed 2021-03-05 by prusnak