trezor/trezor-firmware #1519: production touch panel that appears to not work is fixed by lowering the i2c clockspeed

mcudev, 2021-03-06:
I received a production Trezor T that appeared to have a bad touch panel. The screen displayed fine, I just could not register any touch events. I took the case apart and put the screen + ctpm assembly on one of my dev boards so that I could tinker with a dev build of the bootloader (otherwise the production boardloader would not accept my changed bootloader and was stuck not being able to test this) and was able to get it working by simply lowering the clock speed to 200000 at line https://github.com/trezor/trezor-firmware/blob/master/core/embed/trezorhal/touch_T.h#L100

I have not tested whether this lower speed still works with other touch panels that I have. I have not determined what is the upper bound for the speed that this panel will tolerate. I've just been able to reproduce the fact that lowering this speed makes it work and raising it back to the original value makes it not work.

This is probably very interesting for solving user reports and production device returns.
Also, I'm not sure if this panel has a different firmware than other panels that I have. I mention this because my initial test showed that it did, but I need to go back and make sure that I did not screw up my comparison (reference to a firmware hash from a different touch panel than the one fixed by this bug report https://github.com/trezor/trezor-firmware/issues/81#issuecomment-763991498).

prusnak, 2021-03-06:
According to FT6236 datasheet the I2C timing should be between 10 and 400 kHz. Decreasing the clock speed to 200 kHz probably won't hurt anyone. Fixed in efac5d5

closed 2021-03-06 by prusnak

mcudev, 2021-03-06:
After some testing, this particular panel fails between 280 and 285 kHZ.
Other touch panels that I have seem to handle the lower 200 kHZ fine.
The firmware hash was different. But, after comparing the firmware dump to a few other panels, it differs in only 4 bytes from another panel and those bytes appear to always be different, and so is probably just a serial number for the CTPM's MCU (just a guess).
In summary, this change looks good to me so far. Thanks!

prusnak, 2021-03-06:
Thank you too for suggestion and testing!

commit efac5d56 (2021-03-06): fix(core): change touch i2c clock speed to 200 kHz

fixes #1519