trezor/trezor-firmware #1526: Consider implementing BIP67 (sortedmulti)

prusnak, 2021-03-10:
[BIP67](https://github.com/bitcoin/bips/blob/master/bip-0067.mediawiki) defines that keys within MultisigRedeemScriptType should be sorted. We currently don't do this, because the adoption of such standard was quite low when we were implementing multisig.

We might want to add it now to fully support `sortedmulti` in Bitcoin descriptors.

Related issue: https://github.com/bitcoin-core/HWI/issues/487

Lord-Of-Gamers, 2021-03-15:
approved

prusnak, 2021-10-13:
We could implement this by adding `MultisigRedeemScriptType.sort_final_pubkeys` bool field and let the client decide whether they want to follow BIP67 for the final keys (and possibly break the order given in `MultisigRedeemScriptType.nodes`) or they want to keep the order as is.

somebodyLi, 2022-04-20:
>  I have not looked into core code yet.

In core, implement this is quite simple