trezor/trezor-firmware #1527: Request for silent wiping feature

DWShuo, 2021-03-15:
**Describe the solution you'd like**
Allow the user to silently wipe their device. 
Upon the user entering the wipe code:
1. Trezor wipes all private data with no message displayed
2. New private keys are generated
3. Old wipe code is set as the new pin 
4. The device is unlocked

This gives the appearance that the device was successfully unlocked.

**Additional context**
This feature would be extremely useful when one is required to give a pin under duress. 
Giving away that the device has been wipe would not be less than ideal in such a situation.

matejcik, 2021-03-15:
The device is unlocked into a completely empty wallet with no transaction history. For the affected user in question, this leads to (a) suspicion that the wallet is hidden under a passphrase, which the user can't provide seeing as no passphrase exists, or (b) suspicion that silent wipe was used if the assailant is aware of the existence of such feature.
Neither variant fares any better for the user than unlocking with wipe code.

If a user expects to find themselves in such situation, there's several steps they _should_ take instead of relying on wipe code or the hypothetical silent wipe:
1) openly and publicly set up a multisig scheme that does not allow this single person to access the funds, or
2) set up a series of decoy hidden wallets under distinct passphrases, so that you can provide one under duress and plausibly claim that this is your main account

prusnak, 2021-03-15:
@DWShuo We discussed the silent option in the product meeting when the wipe code was being implemented and came to the conclusion this would cause much more harm than good.

closed 2021-03-15 by prusnak