trezor/trezor-firmware #1528: Support for BIP-85 in Trezor firmware

8go, 2021-03-15:
**Is your feature request related to a problem? Please describe.**
No, this feature request is not related to an existing problem. 
This is a request for new, additional functionality, adding a new useful feature to Trezor devices.

**Describe the solution you'd like**
BIP-85 https://github.com/bitcoin/bips/blob/master/bip-0085.mediawiki allows the derivation
of child seeds from a master seed. This is very useful and should be implemented inside the firmware
of the Trezor device so that the user never has to enter the master seed on any keyboard, etc.

The sample use case is the following: A Trezor user already has a Trezor and a (master) seed, e.g. 
24 words. The user already stores this (master) seed safely on steel plates in 3 different countries. 
Now the user wants to give his teenage son a new Trezor and the user of course wants that the son
uses a different seed, but the user does not want to go through the hassle of creating a brand new seed
and storing it again on new steel plates and traveling again to 3 countries to hide the additional plates. 

BIP-85 comes to the rescue. The user uses his (master) seed to derive a new seed for his son. 
The user does *not* need to create any backup, does *not* need to buy any more steel plates, 
does not need to store any new steel plates, etc. 
Anytime in the future should the user want to know the seed of his son the user can deterministically
derive the child seed again from his master seed via BIP-58 (on the Trezor firmware).
As a matter of fact, the user
can derive an unlimited number of independent child seeds from his master seed.
E.g. the user could derive 12 different child seeds, one for each of his 12 children.

How would that work. 
- The user purchases a new Trezor device. 
- The user installs his master seed on this new Trezor device.
- The user goes into the "derive child seed" (the requested new feature) option in the Trezor software
- The Trezor firmware uses BIP-85 to derive a new child seed. 
- There might be questions like: 
  - how many words do you want? 12? 18? 24?
  - in which language should the words be? English default, spanish, french, chinese, etc.
  - which child index do you want to use? default 0 (first child seed), 1 (2nd child seed), etc.
- Once the Trezor firmware has computed the new child seed, the firmware might ask
  - what do you want to do with this new child seed?
  - Just display it? (A)
  - Replace the current seed with the new child seed? (B)

If the user selects option (B), asks firmware to replace current seed with newly computed child seed
then the device reconfigures itself to store and use the new child seed. 
The user can now give this new Trezor device to its son. 

Other uses cases are:
- a user can derive child seeds for his elderly parents who are not capable of storing 
  and protecting a seed
- a user can derive a child seed for itself, maybe to use one seed for one account, 
  another seed for another account
- a company could use different seeds for different departments,
- etc.

**Describe alternatives you've considered**
Of course one can use BIP-85 today on an offline computer to derive a child seed
via some Python or JavaScript software that implements BIP-58. 
You enter the master seed on the keyboard and the software spits out a child seed.
But that is a lot of work (set up an offline computer, etc.) and not so safe (you have to enter the master seed
on the keyboard or similar). 

This is NOT ideal. The ideal case is that the master seed never leaves the Trezor, and that
one never has to enter the master seed on a keyboard or similar. The only good solution is
that the firmware computes the child seed using BIP-85.

**Additional context**

- BIP-85: https://github.com/bitcoin/bips/blob/master/bip-0085.mediawiki
- Python reference implementation: https://github.com/ethankosakovsky/bip85
- JavaScrpt reference implementation: https://github.com/hoganri/bip85-js
- ColdCard has already implemented this: https://github.com/Coldcard/firmware/pull/39

So, in order to stay competitive and to provide users with useful features, Trezor should also kindly implement BIP-85 in its firmware. This helps to sell more Trezors, and helps Trezor users stay safe.

PS: And yes, it is understood that in the given example, the son must trust the parent. 
The parent can rob the son.
But the parent does not need to trust the son. The son cannot rob the parent.

hommadi2001, 2022-11-09:
This feature would be extremely useful.
It is difficult to support all sorts of coins out there. And frustrating to wait for new coins to be added.
This feature would allow backing up a very wide range of coins using one and only well kept master key.
Please give this a priority kick.

prusnak, 2023-04-11:
> how can people who don't work at Satoshi Labs help this become part of the firmware?

Trezor is an open-source project hosted on GitHub. Anyone can open a pull request which will get reviewed by our firmware maintainers.

mgoelz, 2026-09-02:
I've opened #7786  implementing BIP-85 in core firmware, trezorlib and trezorctl: BIP-39 (12/15/18/21/24 words), HD-Seed WIF, XPRV, HEX and base64/base85 password applications, with optional display of the derived secret on the device and an `on_device_only` mode so a child mnemonic never leaves the device. Feedback on the message design and the UX is welcome.