trezor/trezor-firmware #1530: Add a warning to trezorctl when safety-checks are used for T1

tsusanka, 2021-03-17:
`trezorctl` returns ` Error: DataError: No setting provided` when `trezorctl set safety-checks prompt` is used on T1. We could add a check to trezorctl and inform the user this does not work for legacy. On the other hand this will get significantly less important once Port is done so :shrug:.

OS: Win 10
FW: 1.9.4

matejcik, 2021-03-17:
there's several ways to approach this. One is to hard-code versions that are affected (safety-checks are not in capabilities), which would tie into #1062, but in a naive version that makes the code ugly.

In a somewhat less naive version, we could have trezorctl-side capability database.

But it is probably overkill for this issue. The good enough solution looks like:
```
try:
	call(ApplySettings(...))
except DataError as e:
	if e.msg == "No setting provided":
		raise DataError(f"Unsupported setting value: {value}") from e
```

matejcik, 2021-09-07:
safety-checks are now supported on T1 so this is no longer relevant

closed 2021-09-07 by matejcik