trezor/trezor-firmware #1534: Re-enable secp256k1-zkp and make it default

prusnak, 2021-03-18:
secp256k1(-zkp) performs better for signature verification and contains schnorr algo

prusnak, 2021-03-18:
I did some experiments in the [secp256k1-zkp](https://github.com/trezor/trezor-firmware/tree/secp256k1-zkp) branch

Currently, `trezor.crypto.curves.secp256k1_zkp` is not a drop-in replacement for `trezor.crypto.curves.secp256k1`, so we need to replace `secp256k1.foo()` with `secp256k1_zkp.Context().foo()`. 

We'll need to discuss whether this is something that we want - maybe we should make it nicer by changing this into context manager:

```
with secp256k1_zkp.Context() as secp256k1:
    secp256k1.foo()
```

or we whether we want to allocate the secp256k1_zkp context outside of Python code control and we can do this in two different ways:
* allocate before uPy interpreter runs and keep it resident (outside of uPy heap)
* allocate it within uPy on first use and keep it resident (inside of uPy heap)

onvej-sl, 2021-04-12:
# TLDR
  * Signing, private key to public key derivation and ecdh computation are resistant to simple power analysis (and timing attacks). If we use context randomization (`secp256k1_context_randomize`), signing and private key to public key derivation should be also resistant to differential power analysis. Nevertheless, the randomization makes the operations twice as long. It seems to me that ecdh computation is prone to differential power analysis. All these operation should be resistant to both simple power analysis and differential power analysis in `trezor-crypto`
  * See the tables below for the choice of constants `ECMULT_WINDOW_SIZE` and `ECMULT_GEN_PREC_BITS`.
  
# Suggestion
  * It doesn't make sense to recreate the context before every signing and verification, since
    * creating the context takes longer than signing or verification and
    * the context is modified only by `secp256k1_context_randomize` (or `secp256k1_ecmult_gen_blind`) which serves as protection against differential power analysis and the only part that is modified is `initial` and `blind` in `secp256k1_ecmult_gen_context`.
  * Instead of that I suggest having a global context that is created once. If we want to have a differential power analysis protection, the context should be randomized with `secp256k1_context_randomize` before every call of `secp256k1_ecdsa_sig_sign` and `secp256k1_ec_pubkey_create`. If we want these operations to be thread-safe we may create a "shallow copy" of the context with `memcpy(ctx_clone, ctx, sizeof(secp256k1_context))` before the randomization. It should be OK since `initial` and `blind` are stored in the context as values, whereas all the big tables are stored as pointers.

  * We can also make use of `secp256k1_ec_pubkey_create`, which is faster than `ecdsa_get_public_key{33,65}` (and `scalar_multiply`). If we want to use it even in `bip32.c`, the context cannot be stored on the micropython heap.

# Report
## Constants
### `ECMULT_WINDOW_SIZE`
The constant is an integer from 2 to 24, inclusive. The bigger the constant is
  * the faster is `secp256k1_ecmult` (and `secp256k1_ecdsa_sig_verify`)
  * the bigger is the size of `secp256k1_ecmult_context` (and `secp256k1_context`)
  * the slower is `secp256k1_ecmult_context_build` (and `secp256k1_context_preallocated_create`)

| `ECMULT_WINDOW_SIZE` | `secp256k1_ecdsa_sig_verify` | `secp256k1_ecmult_context_build` | `secp256k1_ecmult_context` |
|---------------------:|-----------------------------:|---------------------------------:|---------------------------:|
|                    2 |                      16.1 ms |                           5.9 ms |                      128 B |
|                    4 |                      14.3 ms |                           6.4 ms |                      512 B |
|                    6 |                      13.5 ms |                           8.3 ms |                       2 kB |
|                    8 |                      12.9 ms |                          16.2 ms |                       8 kB |
|                   10 |                      12.6 ms |                            47 ms |                      32 kB |

### `ECMULT_GEN_PREC_BITS`
The constant is 2, 4 or 8. The bigger the constant is
  * the faster is `secp256k1_ecmult_gen` (and `secp256k1_ecdsa_sig_sign`, `secp256k1_ec_pubkey_create`)
  * the bigger is the size of `secp256k1_ecmult_gen_context` (and `secp256k1_context`)
  * the slower is `secp256k1_ecmult_gen_context_build` (and `secp256k1_context_preallocated_create`), provided the context is not precomputed in compile-time using the flag `USE_ECMULT_STATIC_PRECOMPUTATION`.

| `ECMULT_GEN_PREC_BITS` | `secp256k1_ecdsa_sig_sign` | `secp256k1_ec_pubkey_create` | `secp256k1_ecmult_gen_context` |
|-----------------------:|---------------------------:|-----------------------------:|-------------------------------:|
|                      2 |                    23.4 ms |                      14.4 ms |                          32 kB |
|                      4 |                    11.7 ms |                       7.2 ms |                          64 kB |
|                      8 |                     5.9 ms |                       3.6 ms |                         512 kB |

Only the times for `ECMULT_GEN_PREC_BITS = 4` in the table were measured, other times are estimated.

### `WINDOW_A`
The constant has a fixed value of 5 which is optimal in terms of speed of functions `secp256k1_ecmult` (and `secp256k1_ecdsa_sig_verify`) and `secp256k1_ecmult_const` (and `secp256k1_ecdh`). The bigger the constant is the more memory on stack these functions needs. 

| `WINDOW_A` |        | `secp256k1_ecdsa_sig_verify` |        | `secp256k1_ecdh` |
|-----------:|-------:|-----------------------------:|-------:|-----------------:|
|          2 | 1440 B |                      14.7 ms |  168 B |          23.7 ms |
|          3 | 1772 B |                      13.6 ms |  336 B |          15.1 ms |
|          4 | 2436 B |                      13.1 ms |  672 B |          12.7 ms |
|          5 | 3764 B |                      12.9 ms | 1344 B |          11.8 ms |
|          6 | 6420 B |                      13.2 ms | 2688 B |          12.2 ms |

The times were measured for `ECMULT_GEN_PREC_BITS = 4` and `ECMULT_WINDOW_SIZE = 8`, bytes are a lower estimate.

## Contexts
### `secp256k1_context`
The context consists of `secp256k1_ecmult_context` and `secp256k1_ecmult_gen`.

### `secp256k1_ecmult_gen_context`
[... cut ...] #L186-L197), whereas `secp256k1-zkp` does not.
* `trezor-crypto` rejects signatures of zero digest as a protection against a fault injection attack since its easy to forge a signature of zero digest (see https://github.com/trezor/trezor-firmware/blob/master/crypto/ecdsa.c#L1044-L1051), whereas `secp256k1-zkp` does not.
* It seems to me that `secp256k1_ecmult_const` (and `secp256k1_ecdh`) in `secp256k1-zkp` doesn't use any kind of randomization as a protection against differential power analysis.

prusnak, 2021-04-12:
@onvej-sl WOW, great work Ondrej! Thank you for the very thorough analysis!

prusnak, 2021-06-12:
Just rebased `secp256k1-zkp` on top of the current master

onvej-sl, 2021-11-19:
Implemented in https://github.com/trezor/trezor-firmware/pull/1897 and https://github.com/trezor/trezor-firmware/pull/1678.

closed 2021-11-19 by onvej-sl

invd, 2021-11-21:
I've started integrating the new mainlined zkp functionality into the T1 fuzzer and noticed some API issues. I'm not sure if this ticket is the right place to bring it up, so feel free to move it or respond to it elsewhere.  (#1864 feels like a different scope, though.)

* `zkp_context.h` doesn't expose `bool zkp_context_is_initialized(void)`, so other code can't check the initialization status of the internal `static secp256k1_context *context;` variable
* On regular builds without `-DNDEBUG`, the `assert(context == NULL);` in `int zkp_context_init()` prevents re-initializing non-null contexts
* This presents practical issues since `void zkp_context_destroy()` only clears the context, but does not set `context = NULL;` again

The most straightforward approach is to expose the initialization check and actually null the context pointer, but I haven't looked into this very deeply and may be missing some logic or side effects.

onvej-sl, 2021-11-26:
> I'm not sure if this closed ticket is the right place to bring it up, so feel free to move it or respond to it elsewhere

Next time please open a new issue. It is easier to track it.

invd, 2021-11-27:
> Next time please open a new issue.

Will do, thanks!