trezor/trezor-firmware #1537: Port bootloader's vendor headers to T1

tsusanka, 2021-03-19:
Idea from Asana backlog. It will probably be too much engineering but I am still curious what it takes and whether it is even possible.

tsusanka, 2021-03-29:
Unless we decide to change the memory layout on T1 (so we can have bigger bootloader on T1) we can't do this.

prusnak, 2021-05-17:
Checklist:
- [ ] Ed25519 COSI signatures
- [ ] parse VERSION and FIX_VERSION from the firmware header
- [ ] warn when firmware downgrade is happening
- [ ] warn when firmware downgrade with erase (new VERSION < old FIX_VERSION) is happening
- [ ] erase the storage when new VERSION < old FIX_VERSION

andrewkozlik, 2021-05-17:
- [ ] Add MAX_BOOTLOADER_VERSION to firmware header, which will correspond with the maximum bootloader version listed in bl_check.c.
- [ ] Don't allow user to downgrade if BOOTLOADER_VERSION > MAX_BOOTLOADER_VERSION, because it makes the firmware inoperable.

Hannsek, 2023-03-28:
I assume we can close this as not planned, right?

closed 2023-11-20 by Hannsek